ros-infrastructure/rosdep #367: rosdep fails if pip is not installed

mitchellwills, 2015-01-06:
When installing the base ros system nothing appears to depend on pip. This leads to issues when trying to use rosdep to install dependencies. I have experienced this issue on hydro 12.04 and indigo 13.10. I get the following error message when running rosdep
```
ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.0

Bad installer [pip]: [Errno 2] No such file or directory
```

It seems like it may be nice to make pip be installed implicitly on systems that support it when rosdep will attempt to install using pip.

wjwwood, 2015-01-06:
Could you provide the command you ran to produce this error?

mitchellwills, 2015-01-06:
Woops, sorry. I ran the following in the root of my catkin workspace
```
rosdep install -i --from-paths src
```

It happened to be on a system that I think had some base installation of ROS and that was it. In the src directory there was a package that depended on a rosdep key that resolved to a pip package.

wjwwood, 2015-01-06:
Ok, well in the case that a `pip` key is needed it will assert that `pip` is available, that is intended behavior. If that was the case then I'll close this for now. Feel free to comment further, I can reopen the issue if something is still left to be resolved.

You can add `--simulate` to get what it would try to install, that shouldn't except if `pip` is not installed.

closed 2015-01-06 by wjwwood

mitchellwills, 2015-01-06:
Ok it just seemed like it was weird that it raises an internal error. I would have expected it to just print bad installer (without the internal error message cause it's not really an internal error). Not a huge problem, especially since rosdep will be replaced soon.

wjwwood, 2015-01-06:
I guess I was mistaken, it will except if you use `--simulate`, but it will not when nothing in your workspace depends on a key which resolves to `pip`. I just tested it by getting `ros_base`'s source and uninstall `pip`, then adding the `python-cpplint` depend to catkin's `package.xml`.

wjwwood, 2015-01-06:
Yeah, unfortunately that whole top part is attached to all exceptions. We should probably change the exception handling up a bit to make it more palatable.

taegoobot, 2017-12-28:
Is this issue fixed? I am getting the same message as below by typing the command

`rosdep install --from-paths src --ignore-src --rosdistro kinetic -y`

```
ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.8

Bad installer [pip]: [Errno 2] No such file or directory: 'pip': 'pip'
```

NikolausDemmel, 2017-12-28:
> Is this issue fixed?

Not sure what you mean. Did you read the responses? The issue should be the same as always, i.e. it would be great if the error message read differently.

As mentioned above there should be an easy workaround: Install pip.

aniroodh-ravikumar, 2018-04-04:
> As mentioned above there should be an easy workaround: Install pip.

I have the same issue as @taegoobot , and the issue seems to persist even after pip is installed.

NikolausDemmel, 2018-04-04:
@aniroodh-ravikumar: What happens when you run `pip --version` in the same terminal from which you run `rosdep`? What OS are you on? Can you paste the output of the rosdep command you run and also the output when you run with the `-s` (simmulate) flag?

NikolausDemmel, 2018-04-04:
Yes, you need to make sure that `pip` is runnable from the shell.

Could you run `which python` and `python --version`.

aniroodh-ravikumar, 2018-04-04:
> aniroodhs-mbp:ros_catkin_ws aniroodhravikumar$ which python
/opt/local/bin/python
aniroodhs-mbp:ros_catkin_ws aniroodhravikumar$ python --version
Python 2.7.14

NikolausDemmel, 2018-04-04:
What kind of setup is this that you have python in `/opt/local/bin` on a mac? With homebrew it should be in `/usr/local/bin` or with system python in `/usr/bin/python`, I guess?

In any case, you should be able to upgrade / reinstall pip with `python -m pip install --force-reinstall -U pip`, but you need check where the executables are installed and make sure that directory is on your path...

wjwwood, 2018-04-05:
`/opt/local/bin` is probably macports, which none of us use regularly.