ros-infrastructure/rosdep #368: rosdep cannot detect installation of [argparse]

jmtatsch, 2015-01-22:
rosdep install --from-paths src --ignore-src --rosdistro indigo -y                             tatsch@79

executing command [sudo pip install -U argparse]
The directory '/Users/tatsch/Library/Logs/pip' or its parent directory is not owned by the current user and the debug log has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want the -H flag.
The directory '/Users/tatsch/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want the -H flag.
The directory '/Users/tatsch/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want the -H flag.
Requirement already up-to-date: argparse in /Library/Python/2.7/site-packages
ERROR: the following rosdeps failed to install
  pip: Failed to detect successful installation of [argparse]

tfoote, 2015-01-22:
What is your platform? 

Can you run the command it's trying to run manually?

Can you make a one line command which does not use the source directory or provide us with a rosinstall for your source directory for us to reproduce?

jmtatsch, 2015-01-22:
OSX Yosemite / ROS Indigo

If I run the command manually:

````
 sudo pip install -U argparse
Password:
The directory '/Users/tatsch/Library/Logs/pip' or its parent directory is not owned by the current user and the debug log has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want the -H flag.
The directory '/Users/tatsch/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want the -H flag.
The directory '/Users/tatsch/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want the -H flag.
Requirement already up-to-date: argparse in /usr/local/lib/python2.7/site-packages
````

Rosinstall is here:

https://gist.github.com/jmtatsch/a4ff114be8378e5f32c4

tfoote, 2015-01-22:
Looking closer the first 3 lines appear to be warnings then it says the requirement is already up to date. But it appears to be returning a non-zero return code. 

This probably has to do with argparse being promoted to be part of the core, and pip. 

Here it's requred for all platforms: https://github.com/ros/rosdistro/blob/master/rosdep/osx-homebrew.yaml#L453

But for any platforms with 2.7 installed we don't need to ask for anything specific since it's built in. We can use [] to indicate nothings is requred to install as we do in the Ubuntu rules: https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L74-L105

The trick will be to differentiate where osx has 2.6 vs 2.7 versions of python.

cc/ @wjwwood

wjwwood, 2015-01-22:
I opened a pull request for the warnings from `pip`, I'm not sure about the `argparse` thing. I can reproduce the problem, but I'm not sure what has broken in `rosdep` or changed in `pip` to cause this.

wjwwood, 2015-01-22:
Looks like after updating `pip` the `pip freeze` command no longer reports `argparse` as installed. This makes sense in someways, but I'm still trying to find out what else it might be excluding.

I guess the right thing to do would be to update the rosdep rules, but I would also consider having a special case for `argparse` in the `pip` detector.

wjwwood, 2015-01-22:
This is a new thing in `pip`, I commented on the upstream ticket:

https://github.com/pypa/pip/issues/1570#issuecomment-71111030

In the short term I guess we can add a rule for `argparse` (it will be hard to tell by the OS X os version whether or not the right Python is being used). We cannot even assume that the Python that `rosdep` is running in is the same one it is installing for (imagine it is in a `virtualenv` and the `pip` it is using is for the system).

A workaround for users until we get a fix in `rosdep` is to use `--skip-keys python-argparse`.

jmtatsch, 2015-01-23:
`--skip-keys xxx` is a godsend. Have been looking for that for a long time ;)

wjwwood, 2015-01-27:
I think this is addressed with #371

closed 2015-01-27 by wjwwood

jmtatsch, 2015-02-11:
After a rosdep update, the same thing happened again.

wjwwood, 2015-02-12:
@jmtatsch what version of `rosdep`? Can you provide output?

wjwwood, 2015-02-13:
@jmtatsch I was unable to reproduce this on my machine with rosdep `0.11.1`, can you confirm this is the case for you?

jmtatsch, 2015-02-13:
Dunno why it didn't work. Its working now with 0.11.1.

jjkllll, 2017-06-21:
I have installed googlemaps, numpy and polyline using pip install command. When i run the code as sudo python code.py it works. But, when i use rospy in code and type rosrun package code.py it fails. I want to add these installed python libraries on ROS so that i can run the code on ROS.

PR #371 (2015-01-27): pip: fallback to pip show to detect things like argparse
This is designed to address #368 based on the feedback https://github.com/pypa/pip/issues/1570#issuecomment-71166391. This is the workaround until they decide to change the behavior of `pip show` or until they provide some other mechanism to detect these things.

Fixes #368