ros-infrastructure/rosdep #372: Fedora Traceback

DLu, 2015-02-08:
dlu@santo:~$ rosdep update
    reading in sources list data from /etc/ros/rosdep/sources.list.d
    Hit https://github.com/ros/rosdistro/raw/master/rosdep/osx-homebrew.yaml
    Hit https://github.com/ros/rosdistro/raw/master/rosdep/gentoo.yaml
    Hit https://github.com/ros/rosdistro/raw/master/rosdep/base.yaml
    Hit https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml
    Hit https://github.com/ros/rosdistro/raw/master/rosdep/ruby.yaml
    Hit https://github.com/ros/rosdistro/raw/master/releases/fuerte.yaml
    Ignore legacy gbpdistro "groovy"
    Query rosdistro index https://raw.github.com/ros/rosdistro/master/index.yaml
    Add distro "groovy"
    Add distro "jade"
    
    ERROR: Rosdep experienced an error: 'fedora'
    Please go to the rosdep page [1] and file a bug report with the stack trace below.
    [1] : http://www.ros.org/wiki/rosdep
    
    rosdep version: 0.10.25
    
    Traceback (most recent call last):
      File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 121, in rosdep_main
        exit_code = _rosdep_main(args)
      File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 264, in _rosdep_main
        return _no_args_handler(command, parser, options, args)
      File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 272, in _no_args_handler
        return command_handlers[command](options)
      File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 437, in command_update
        error_handler=update_error_handler)
      File "/usr/lib/pymodules/python2.7/rosdep2/sources_list.py", line 436, in update_sources_list
        rosdep_data = get_gbprepo_as_rosdep_data(d)
      File "/usr/lib/pymodules/python2.7/rosdep2/gbpdistro_support.py", line 175, in get_gbprepo_as_rosdep_data
        APT_INSTALLER: {'packages': [deb_package_name]}
    KeyError: 'fedora'

cottsay, 2015-02-08:
I can't find that line in the current version of rosdep, so that looks like an outdated version.

Probably a duplicate of https://github.com/ros-infrastructure/rosdep/issues/359

DLu, 2015-02-08:
Now I feel silly. I did search for the bug, but apparently only in open tickets. Problem solved.

closed 2015-02-08 by DLu