ros-infrastructure/rosdep #373: rosdep "abuses" pip on OS X 10.10.2

jmtatsch, 2015-02-13:
```
rosdep install --from-paths src --ignore-src --rosdistro indigo -y                                                              
Usage: pip COMMAND [OPTIONS]
No command by the name pip show
  (maybe you meant "pip install show")
Usage: pip COMMAND [OPTIONS]
No command by the name pip show
  (maybe you meant "pip install show")
Usage: pip COMMAND [OPTIONS]
No command by the name pip show
  (maybe you meant "pip install show")
Usage: pip COMMAND [OPTIONS]
No command by the name pip show
  (maybe you meant "pip install show")
Usage: pip COMMAND [OPTIONS]
No command by the name pip show
  (maybe you meant "pip install show")
Usage: pip COMMAND [OPTIONS]
No command by the name pip show
  (maybe you meant "pip install show")
```

then the installation of the dependencies continues as expected. But I guess the pip packages are skipped.

wjwwood, 2015-02-13:
What system is this? That's a older version of pip, i.e. `pip<1.2.1post1`:

http://stackoverflow.com/questions/14060043/why-dont-pip-show-or-pip-list-work-for-me

jmtatsch, 2015-02-13:
OS X Yosemite. Sorry, must have overread the part that 
`sudo easy_install pip==1.2.1`
is only for mavericks.

closed 2015-02-13 by jmtatsch