ros-infrastructure/rosdep #375: rosdep ignores --skip-keys

jmtatsch, 2015-02-19:
e.g.
```rosdep install --from-paths src --ignore-src --rosdistro indigo -y --skip-keys "ogre gazebo"
executing command [brew install ogre]```

wjwwood, 2015-02-19:
Annoyingly, it is one key at a time, do this:

```
rosdep install --from-paths src --ignore-src --rosdistro indigo -y --skip-keys ogre --skip-keys gazebo
```

dirk-thomas, 2015-02-19:
The help text suggests that a single space separated string would be fine as well: https://github.com/ros-infrastructure/rosdep/blob/1b1efed99a14f3f17ea9dcff44c9d65c0b81d675/src/rosdep2/main.py#L289-L298

wjwwood, 2015-02-19:
I think it was added by @NikolausDemmel in https://github.com/ros-infrastructure/rosdep/commit/e6afc9f53adfc72cc83d8e438a1ae972e2f7e789, but I'm pretty sure that was not the intended behavior. I don't have time to fix this right now, but I would take a pull request.

dirk-thomas, 2015-02-19:
@jmtatsch I tried a similar command line and it seems to work just fine. Can you please check which version of rosdep you are using (`rosdep --version`)?

jmtatsch, 2015-02-19:
0.11.1

dirk-thomas, 2015-02-19:
Can you please provide the full output of the command as well as running the same command with the `--skip-keys` but `--simulate` instead.

jmtatsch, 2015-02-19:
rosdep install --from-paths src --ignore-src --rosdistro indigo -y --simulate          
#[homebrew] Installation commands:
  brew install pyside
  brew install gtk+
  brew install gtest
  brew install vtk --with-qt
  brew install log4cxx
  brew install collada-dom
  brew install gazebo
  brew install pcl --HEAD
  brew install pyqwt
  brew install boost-python

rosdep install --from-paths src --ignore-src --rosdistro indigo -y --skip-keys ogre --skip-keys gazebo --simulate
#[homebrew] Installation commands:
  brew install pyside
  brew install vtk --with-qt
  brew install collada-dom
  brew install boost-python
  brew install ogre
  brew install pyqwt
  brew install log4cxx
  brew install gtk+
  brew install pcl --HEAD
  brew install gtest

wjwwood, 2015-02-19:
The `rosdep` key is `libogre-dev`: https://github.com/ros/rosdistro/blob/master/rosdep/osx-homebrew.yaml#L210

You would need to do `rosdep install --from-paths src --ignore-src --rosdistro indigo -y --skip-keys libogre-dev --skip-keys gazebo --simulate`.

I still think that I've run into the situation where multiple keys cannot be passed to `--skip-keys`, but I think I would want to verify that before trying to fix it.

jmtatsch, 2015-02-20:
With libogre-dev it works. Thank you.

closed 2015-02-20 by jmtatsch

omerts, 2023-06-07:
@wjwwood Not sure why this was closed, but this is still not working:
Running:
```
rosdep keys --ignore-src --rosdistro=humble --from-paths src --skip-keys='navigation2 librealsense2'  | grep librealsense2
```
Still lists librealsense
[image]