ros-infrastructure/rosdep #377: ERROR: Rosdep experienced an error: XML or text declaration not at start of entity

leeehui, 2015-03-21:
ROS: Indigo OS : Ubuntu14.04

When I run rosdep install turtlesim, the output is:

ERROR: Rosdep experienced an error: XML or text declaration not at start of entity: line 2, column 0
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 345, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 401, in _package_args_handler
    val = rospkg.expand_to_packages(args, rospack, rosstack)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 428, in expand_to_packages
    package_list = rospack.list()
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 188, in list
    self._update_location_cache()
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 180, in _update_location_cache
    list_by_path(self._manifest_name, path, cache)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 67, in list_by_path
    root = ElementTree(None, os.path.join(d, PACKAGE_FILE))
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 611, in __init__
    self.parse(file)
  File "<string>", line 38, in parse
ParseError: XML or text declaration not at start of entity: line 2, column 0

please help! thanks.

wjwwood, 2015-03-24:
@leeehui it looks like you may have an invalid `manifest.xml` on your `ROS_PACKAGE_PATH` somewhere.

leeehui, 2015-03-28:
@wjwwood thanks for your answer. finally I found that I add two lines in .bashrc of my home space as follow:
"source /opt/ros/indigo/setup.bash"
"source ~/catkin_ws/devel/setup.bash"
this is not right.
I just executed source /opt/ros/indigo/setup.bash  in shell 
then ran rosdep install turtlesim,  and it worked well.

so, I think the reason is that I don't actually understand what the setup.bash do

closed 2015-03-28 by leeehui