ros-infrastructure/rosdep #378: rosdep permissions

Mdu7078, 2015-04-03:
So the permissions were messed up with rosdep and it suggested to "Try running 'sudo rosdep fix-permissions'" to fix this, but when I did so the following error was produced on a machine running Ubuntu 14.04:

ERROR: Rosdep experienced an error: 'getgrgid(): gid not found: 15318'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 343, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 351, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 769, in command_fix_permissions
    group_name = grp.getgrgid(gid).gr_name
KeyError: 'getgrgid(): gid not found: 15318'

wjwwood, 2015-04-03:
Thanks for the report, I'll have a look into the issue asap.

@Mdu7078 As a work around you can probably try to remove the folder `~/.ros/rosdep` and then do `rosdep update`.

@NikolausDemmel I believe you contributed this verb for rosdep, does anything jump out at you here that might help me narrow down the issue? (I haven't looked into it at all yet)

tfoote, 2015-04-03:
One things that sticks out is that the group id is very high. Below 1000 is reserved and typically new groups start being numbered at 1000 and incrementing upward.

NikolausDemmel, 2015-04-03:
Could this be to do with LDAP groups? Are you in a corporate net (or similar) with LDAP authentication?

NikolausDemmel, 2015-04-04:
@Mdu7078 Could you please post the output if running `id` on the command line?

Mdu7078, 2015-04-06:
I am running this on a server that has a very large number of users.

This is the output of `id`:
uid=15318(mdu7078) gid=3128(domain users) groups=3128(domain users)

Mdu7078, 2015-04-06:
@wjwwood Removing `~/.ros/rosdep` and then runing `rosdep update` did not work as a workaround and instead produces this error:

```
ERROR: Rosdep experienced an error: [Errno 13] Permission denied: '/home/mdu7078/.ros/rosdep/sources.cache'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 343, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 351, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 533, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 453, in update_sources_list
    retval.append((source, write_cache_file(sources_cache_dir, source.url, rosdep_data)))
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 531, in write_cache_file
    os.makedirs(source_cache_d)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/home/mdu7078/.ros/rosdep/sources.cache'
```

I even modified the permissions to `~/.ros/` (which is the only folder that exists currently) so that it should be accessible and this still doesn't work.

wjwwood, 2015-04-07:
@Mdu7078 I don't know then, all `rosdep` tries to do is create the folder and the `sources.cache` file, inheriting any permissions from the shell it is run in. This sounds like something your system administrator will need to address. You can try removing the entire `~/.ros` folder.

Mdu7078, 2015-04-07:
That worked.  I guess for whatever reason, `~/.ros` had elevated permissions.

wjwwood, 2015-04-07:
Maybe I'm missing something, but it looks like your user id is `15318` and that is what is being passed to `grp.getgrgid`? If that is the case, then it doesn't look quite right. Wouldn't pass the group id to that function instead?

wjwwood, 2015-04-07:
Hmm, could it be that your home directory, `~`, (incorrectly) had a group id which was the same as your user id? We get the user id and group id by `stat`'ing the home directory:

https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/main.py#L775-L779

Mdu7078, 2015-04-07:
This is the output of `stat ~/` 
```
  File: ‘/home/mdu7078/’
  Size: 4096      	Blocks: 8          IO Block: 32768  directory
Device: 1ah/26d	Inode: 2690041785  Links: 1
Access: (0700/drwx------)  Uid: (15318/ mdu7078)   Gid: (15318/ UNKNOWN)
Access: 2015-04-06 14:03:11.526900000 -0400
Modify: 2015-04-06 20:10:04.958569000 -0400
Change: 2015-04-06 20:10:04.958569000 -0400
 Birth: -
```
Just like you predicted, my group id is the same as my user id.

wjwwood, 2015-04-07:
Yeah, so that seems not right, but it's hard for me to say if it's supposed to be that way or not. You might need to reach out to your sys admin.

I guess the best thing we can do is catch this error and report the group as unknown just like stat is.

NikolausDemmel, 2015-04-08:
For `fix-permissions`, the group name is actually only used for command line output. We could robustify the code by falling back to printing just uid / gid if the names cannot be determined.

spaghetti-, 2015-08-18:
I added a check for groupid as I believe username will always be available?

wjwwood, 2015-08-18:
Fixed by https://github.com/ros-infrastructure/rosdep/pull/399

closed 2015-08-18 by wjwwood