ros-infrastructure/rosdep #379: Python 3 rosdep database is incompatible with Python 2

cottsay, 2015-04-06:
I (accidentally) used a Python 3 version of rosdep to run `rosdep update`. I then tried releasing a package in a Python 2 version of Bloom, which called into the rosdep API, and threw this:
```
Traceback (most recent call last):
  File "/usr/bin/git-bloom-generate", line 9, in <module>
    load_entry_point('bloom==0.5.19', 'console_scripts', 'git-bloom-generate')()
  File "/usr/lib/python2.7/site-packages/bloom/commands/git/generate.py", line 252, in main
    run_generator(generator, args)
  File "/usr/lib/python2.7/site-packages/bloom/commands/git/generate.py", line 127, in run_generator
    gen.pre_modify)
  File "/usr/lib/python2.7/site-packages/bloom/commands/git/generate.py", line 101, in try_execute
    retcode = func(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/bloom/generators/rpm/generator.py", line 540, in pre_modify
    while not self._check_all_keys_are_valid(peer_packages):
  File "/usr/lib/python2.7/site-packages/bloom/generators/rpm/generator.py", line 511, in _check_all_keys_are_valid
    retry=False)
  File "/usr/lib/python2.7/site-packages/bloom/generators/common.py", line 137, in resolve_rosdep_key
    view = get_view(os_name, os_version, ros_distro)
  File "/usr/lib/python2.7/site-packages/bloom/generators/common.py", line 79, in get_view
    value = get_catkin_view(ros_distro, os_name, os_version, False)
  File "/usr/lib/python2.7/site-packages/rosdep2/catkin_support.py", line 108, in get_catkin_view
    sources_loader = SourcesListLoader.create_default(matcher=sources_matcher)
  File "/usr/lib/python2.7/site-packages/rosdep2/sources_list.py", line 607, in create_default
    sources = load_cached_sources_list(sources_cache_dir=sources_cache_dir, verbose=verbose)
  File "/usr/lib/python2.7/site-packages/rosdep2/sources_list.py", line 508, in load_cached_sources_list
    return parse_sources_data(cache_data, origin=cache_index, model=model)
  File "/usr/lib/python2.7/site-packages/rosdep2/sources_list.py", line 366, in parse_sources_data
    raise InvalidData("line:\n\t%s\n%s"%(line, e), origin=origin)
InvalidData: line:
	yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx
unsupported pickle protocol: 4
```

I don't know anything about pickle, but I have to believe that the databases can be made to work with both Python 2 and Python 3...

Thanks,

--scott

wjwwood, 2015-04-06:
@cottsay thanks for the report, I remember @NikolausDemmel dealing with this issue in the Xylem prototype, but I'm not sure when I'll have time to track that down and apply it to `rosdep`. At least this issue is here for others to find now. Thanks again.

dirk-thomas, 2015-04-06:
We noticed this already a while ago when working Python 3 compatibility. The fix should be trivial. The data must be saved with a pickle protocol which all Python version can read. I assume passing the protocol version explicitly here (https://github.com/ros-infrastructure/rosdep/blob/600e9055f3576bc371e90adbb75c3660ee04549d/src/rosdep2/sources_list.py#L535) should make it work.

NikolausDemmel, 2015-04-06:
Yes we should probably be pickling with protocol version 2 (https://docs.python.org/3/library/pickle.html#data-stream-format). Question is if we need the "fix imports" flag (https://docs.python.org/3/library/pickle.html#pickle.dump)?

How can we add unit tests that require both python versions? All I can think of is commiting a small pickled binary blob for each python version and test successful unpickling. However this does not make sure the binary blobs are up to date.

sylvainar, 2016-05-23:
Hello guys,

I'm experiencing this problem too. I've removed every configuration file from rosdep, but I don't know how to force rosdep to execute on Python 2.7. Can you please tell me how you did it ?

Thanks !

NikolausDemmel, 2016-05-24:
As discussed in https://github.com/ros-infrastructure/rosdep/issues/457#issuecomment-221432546, if specifying protocol 2 for both python 2 and 3, there might still be work to do, since string from python 3 deserializes as unicode in python 2, etc...

A first step would be a more informative error message with instructions on how to remove the cache.

Alessandro-Barbieri, 2017-05-25:
see https://bugs.gentoo.org/show_bug.cgi?id=619650

Alessandro-Barbieri, 2017-05-28:
Any progress on this? In gentoo is still real.

BenGHolmes, 2018-09-29:
Any fix for this? Still having issues

jlack1987, 2018-10-28:
Have found my way to this issue as well. Any work arounds? Python 2/3 versioning is a nightmare. I'm not sure how to do a new release of my package

NikolausDemmel, 2018-10-28:
PR #633 implements a fix that is ready to merge (IMHO). You could install rosdep from source on that branch for a workaround.

jlack1987, 2018-10-28:
oh ok, thanks for the update! I'll wait for it and continue once it's merged in.

nuclearsandwich, 2018-11-07:
rosdep 0.13.0 was just released which includes #633. After a `rosdep update` rosdep caches built from either python2 or python3 should interoperate with each other. 

If you have interop issues after updating to 0.13.0 please open a new issue as it likely implies a drift between the data structures used by python2 or python3 or other bug.

closed 2018-11-07 by nuclearsandwich