ros-infrastructure/rosdep #380: rosdep does not pick up changed dependencies in overlayed packages when the same package is installed

stwirth, 2015-04-15:
I have a package installed as debian in `/opt/ros/indigo` and I am working on a source checkout of the same package in my catkin workspace. When I source `/opt/ros/indigo/setup.bash` and then run `cd /path/to/workspace/src/ && rosdep check -v --from-paths .`, the list of required rosdep keys only lists dependencies of the installed package, any changes in the package.xml in the catkin workspace are not reflected. However, when I source `/path/to/workspace/devel/setup.bash` the dependencies are picked up by rosdep.
The problem is that `/path/to/workspace/devel/setup.bash` is not there before running `catkin_make` which in turn fails if the dependencies are not installed.

stwirth, 2015-04-15:
The problem seems to be that the installed package is in the ROS_PACKAGE_PATH and rosdep reads the package.xml from there instead of from the current path.
If I just `unset ROS_PACKAGE_PATH` and then run `rosdep check -v --from-paths .`, the correct `package.xml` is used.

stwirth, 2015-04-15:
```
$ rosdep --version
0.11.2
```

stwirth, 2015-04-15:
Steps to reproduce:
```
 $ source /opt/ros/indigo/setup.bash
 $ mkdir -p test-ws/src && cd test-ws/src && catkin_init_workspace
 $ git clone http://github.com/ros/ros_comm.git
 $ vi ros_comm/clients/roscpp/package.xml
add "<run_depend>my_awesome_package</run_depend>"
 $ rosdep check -v --from-paths ros_comm/clients/roscpp/
...
resolving for resources [roscpp]
resolve_all: resource [roscpp] requires rosdep keys [cpp_common, message_generation, pkg-config, rosconsole, roscpp_serialization, roscpp_traits, rosgraph_msgs, roslang, rostime, std_msgs, xmlrpcpp, catkin, cpp_common, message_runtime, rosconsole, roscpp_serialization, roscpp_traits, rosgraph_msgs, rostime, std_msgs, xmlrpcpp]
...
(my_awesome_package does not show up)
 $ unset ROS_PACKAGE_PATH
 $ rosdep check -v --from-paths ros_comm/clients/roscpp/
...
resolving for resources [roscpp]
resolve_all: resource [roscpp] requires rosdep keys [cpp_common, message_generation, pkg-config, rosconsole, roscpp_serialization, roscpp_traits, rosgraph_msgs, roslang, rostime, std_msgs, xmlrpcpp, my_awesome_package, catkin, cpp_common, message_runtime, rosconsole, roscpp_serialization, roscpp_traits, rosgraph_msgs, rostime, std_msgs, xmlrpcpp]
...
(my_awesome_package shows up)
```

wjwwood, 2015-04-15:
I think you need to add the `--ignore-src` option, which avoids trying to install dependencies it finds on the RPP and in the `--from-paths` directories.

wjwwood, 2015-04-15:
Nvm, this is something else. I can reproduce what you're talking about.

It's likely a side affect of the caching in `rosdep`.

It's still not clear to me what the original issue is, since `catkin_make` doesn't call `rosdep` when building.

Also `catkin_make` adjusts the RPP before running, which can also fix your issue, try this line:

```
$ ROS_PACKAGE_PATH=`pwd`:$ROS_PACKAGE_PATH rosdep check -v --from-paths ros_comm/clients/roscpp/
```

Obviously before you `unset ROS_PACKAGE_PATH`.

stwirth, 2015-04-16:
Ah, great.
```
$ ROS_PACKAGE_PATH=`pwd`:$ROS_PACKAGE_PATH rosdep check -v --from-paths ros_comm/clients/roscpp/
```
works, but this doesn't:
```
ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:`pwd` rosdep check -v --from-paths ros_comm/clients/roscpp/
```
So I think what is wrong is this line:
https://github.com/ros-infrastructure/rosdep/blob/d7481bcbdae73c3b4e11299385a37e48a2a8348b/src/rosdep2/main.py#L401
The path from `from_paths` has to be prepended to ROS_PACKAGE_PATH and not appended so that the overlay works correctly.

wjwwood, 2015-09-24:
Sorry I dropped the ball on this one, but I'm about to do a `rosdep` release and I wanted to get this in:

https://github.com/ros-infrastructure/rosdep/pull/412

PR #412 (2015-09-24): prepend rather than append the paths given by --from-paths to the RPP
This fixes #380 by ensuring that the version of the package in one of the given paths is found before an overlaid copy from the system is found. If you don't ensure this, then new dependencies added to the local copy will not be considered by rosdep, but rather it will use the dependencies of the `package.xml` of the overlaid package from the system.
closed 2015-09-24 by wjwwood