ros-infrastructure/rosdep #381: ERROR: Rosdep experienced an error: 'fedora'

lanshanxunmeng, 2015-04-16:
$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.github.com/ros/rosdistro/master/index.yaml
Add distro "groovy"
Add distro "jade"

ERROR: Rosdep experienced an error: 'fedora'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.10.25

Traceback (most recent call last):
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 121, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 264, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 272, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 437, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/pymodules/python2.7/rosdep2/sources_list.py", line 436, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(d)
  File "/usr/lib/pymodules/python2.7/rosdep2/gbpdistro_support.py", line 175, in get_gbprepo_as_rosdep_data
    APT_INSTALLER: {'packages': [deb_package_name]}
KeyError: 'fedora'

I just follow the website steps,but this error appears,how to resolve ?

wjwwood, 2015-04-16:
You're using a quite old version of `rosdep`, please update, this is the latest version:

```
% rosdep --version
0.11.2
```

closed 2015-04-16 by wjwwood

lanshanxunmeng, 2015-04-16:
oh,Thinks ..