ros-infrastructure/rosdep #387: Gentoo USE flags are not automatically satisfied

allenh1, 2015-07-04:
Portage has a wonderful tool for automatically satisfying USE flags. If you add the flag "--autounmask-write" to an emerge command, then the proper USE flags will be established for the packages you desire. After running emerge with this command, however, one must run "etc-update" to perform the modifications, then one re-runs emerge to begin the actual emerge process. This would greatly simplify the process for Gentoo users, and would make Gentoo installs more Ubuntu-like with respect to rosdep.

wjwwood, 2015-07-08:
I'm not sure what exactly you're asking us to change. Please propose a pull request with a clear description of what's changing and I'll give it a review.

allenh1, 2015-07-08:
Will do.

allenh1, 2015-07-13:
I've decided against this. It's not a good idea to automatically modify use flags like that... I think I'll just make a note about handling it with the installation instructions.

closed 2015-07-13 by allenh1