ros-infrastructure/rosdep #390: object of type 'map' has no len()

nguillermin, 2015-07-24:
Following a `rosdep install --from-paths src --ignore-src --rosdistro jade -y` I get this:

     rosdep version: 0.11.2
     
     Traceback (most recent call last):
       File "/Users/user/.virtualenvs/ros/lib/python3.4/site-packages/rosdep2/main.py", line 136, in rosdep_main
          exit_code = _rosdep_main(args)
       File "/Users/user/.virtualenvs/ros/lib/python3.4/site-packages/rosdep2/main.py", line 345, in _rosdep_main
         return _package_args_handler(command, parser, options, args)
       File "/Users/user/.virtualenvs/ros/lib/python3.4/site-packages/rosdep2/main.py", line 435, in _package_args_handler
         return command_handlers[command](lookup, packages, options)
       File "/Users/user/.virtualenvs/ros/lib/python3.4/site-packages/rosdep2/main.py", line 619, in command_install
         uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
       File "/Users/user/.virtualenvs/ros/lib/python3.4/site-packages/rosdep2/installers.py", line 405, in get_uninstalled
         resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
       File "/Users/user/.virtualenvs/ros/lib/python3.4/site-packages/rosdep2/lookup.py", line 391, in resolve_all
         self.resolve(rosdep_key, resource_name, installer_context)
       File "/Users/user/.virtualenvs/ros/lib/python3.4/site-packages/rosdep2/lookup.py", line 480, in resolve
         resolution = installer.resolve(rosdep_args_dict)
       File "/Users/user/.virtualenvs/ros/lib/python3.4/site-packages/rosdep2/platforms/osx.py", line 300, in resolve
         options = handle_options(options)
       File "/Users/user/.virtualenvs/ros/lib/python3.4/site-packages/rosdep2/platforms/osx.py", line 284, in handle_options
         if len(options) > len(packages):
     TypeError: object of type 'map' has no len()

OS X 10.10.4
Python 3.4 from homebrew

nguillermin, 2015-07-24:
`options` appears to be the problem, as it is the one of type 'map'.

wjwwood, 2015-07-25:
Yeah, looks like this line is the culprit:

https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/platforms/osx.py#L272

`map` returns a generator in Python3, whereas this code expects a list.

wjwwood, 2015-07-25:
@nguillermin can you have a look at https://github.com/ros-infrastructure/rosdep/pull/391 and see if it resolves the issue for you?

nguillermin, 2015-07-25:
It does clear that error. There are more python2/3 errors after that but I think we can close this (and maybe note on the wiki that is Py3 is not supported yet).

Also, an alternate solution which I devised yesterday would be

`options = coerce_to_list(options)`

which I feel makes more sense than endlessly casting things. I'm not sure what `options` is supposed to look like, but mapping `coerce_to_list` to a list you'll end up with a list of lists and I'm wondering if that's really what's intended.

wjwwood, 2015-07-27:
It's probably pretty close to having Python3 support. We do have CI run over it for Python3, but we don't have tests for the UI part of the code.

I'll try to run it with Python3 next week and add all the necessary fixes to #391.

wjwwood, 2015-07-28:
I'm able to run rosdep in a Python3 venv after applying #391, can you post the other issues you were having?

nguillermin, 2015-07-28:
I'm sorry, I fixed the other issues I had and forgot to document what I did. If I have any other issues I'll be sure to open another bug report.

closed 2015-07-28 by nguillermin