ros-infrastructure/rosdep #394: Rosdep experienced an error: Expected file type is 'distribution'

susu3621, 2015-08-06:
The issue happens when I try to make a clean install the ROS on Ubuntu 14.04.2 LTS.
I follow the instructions on http://wiki.ros.org/indigo/Installation/Ubuntu after execute "rosdep update"

Every operation follow the wiki, the only difference is I used the following mirror site.
sudo sh -c '. /etc/lsb-release && echo "deb http://mirrors.ustc.edu.cn/ros/ubuntu/ $DISTRIB_CODENAME main" > /etc/apt/sources.list.d/ros-latest.list'

Hope the information is helpful to target the issue. 

ERROR: Rosdep experienced an error: Expected file type is 'distribution'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 343, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 351, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 533, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 466, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 146, in get_gbprepo_as_rosdep_data
    distro_file = get_release_file(gbpdistro)
  File "/usr/lib/python2.7/dist-packages/rosdep2/rosdistrohelper.py", line 76, in get_release_file
    dist_file = rosdistro.get_distribution_file(get_index(), distro)
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 129, in get_distribution_file
    return create_distribution_file(dist_name, data)
  File "/usr/lib/python2.7/dist-packages/rosdistro/distribution_file.py", line 130, in create_distribution_file
    return DistributionFile(dist_name, data)
  File "/usr/lib/python2.7/dist-packages/rosdistro/distribution_file.py", line 45, in __init__
    assert 'type' in data, "Expected file type is '%s'" % DistributionFile._type
AssertionError: Expected file type is 'distribution'

wjwwood, 2015-08-11:
I'm not sure what that error means, please make sure you have the latest version of `python-rosdistro` and that you've not set the `ROS_DISTRO_URI` in your environment to something else.

wjwwood, 2015-08-11:
I'm also going to close, assuming that updating `python-rosdistro` will resolve the issue for you. But if that's not the case please continue to comment here and we can reopen the issue if necessary.

closed 2015-08-11 by wjwwood