ros-infrastructure/rosdep #395: rosdep install fails on Archlinux

Mizux, 2015-08-06:
Hi,
I'm trying to install ROS (desktop-full) from source
```
rosinstall_generator desktop_full --rosdistro jade --deps --wet-only --tar > jade-desktop-full-wet.rosinstall
wstool init -j8 src jade-desktop-full-wet.rosinstall
yaourt -S python-coverage python-mock python-cairo python-opengl python-nose
yaourt -S python-pygraphviz 
```

But I have the following erros
```
[^_^]─mizux@Flex2%rosdep install --from-paths src --ignore-src --rosdistro jade -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
qt_dotgraph: No definition of [python-pygraphviz] for OS [arch]
rqt_bag_plugins: No definition of [python-cairo] for OS [arch]
rospack: No definition of [python-coverage] for OS [arch]
rosgraph: No definition of [python-mock] for OS [arch]
catkin: No definition of [python-mock] for OS [arch]
roslisp: No definition of [python-coverage] for OS [arch]
rqt_tf_tree: No definition of [python-mock] for OS [arch]
rqt_pose_view: No definition of [python-opengl] for OS [arch]
rqt_dep: No definition of [python-mock] for OS [arch]
[1]─[~/ros_catkin_ws]
[>_<]─aldeb@Flex2%
```

In doubt I still trying to start a compile (adding depencies each time cmake/compile fails)
```
yaourt -S gtest console-bridge poco python-sip  python-distutils-extra
pip install --user empy
./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release
```

ps: Since I have sometime errors I had to install few other dependencies...

Edit: compile failed in 
==> Processing plain cmake package: 'python_orocos_kdl'
==> Building with env: '/home/aldeb/ros_catkin_ws/install_isolated/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/aldeb/ros_catkin_ws/build_isolated/python_orocos_kdl/install'
==> make -j4 -l4 in '/home/aldeb/ros_catkin_ws/build_isolated/python_orocos_kdl/install'
[ 10%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart7.o
/home/aldeb/ros_catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip: In function ‘int convertTo_std_string(PyObject*, void**, int*, PyObject*)’:
/home/aldeb/ros_catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:41:37: error: ‘PyString_Check’ was not declared in this scope
         return (PyString_Check(sipPy) || PyUnicode_Check(sipPy));
                                     ^
/home/aldeb/ros_catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:48:58: error: ‘PyString_AS_STRING’ was not declared in this scope
         *sipCppPtr = new std::string(PyString_AS_STRING(s));
                                                          ^
/home/aldeb/ros_catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:52:30: error: ‘PyString_Check’ was not declared in this scope
      if (PyString_Check(sipPy)) {
                              ^
/home/aldeb/ros_catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:53:62: error: ‘PyString_AS_STRING’ was not declared in this scope
         *sipCppPtr = new std::string(PyString_AS_STRING(sipPy));
                                                              ^
/home/aldeb/ros_catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip: In function ‘PyObject* convertFrom_std_string(void*, PyObject*)’:
/home/aldeb/ros_catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/std_string.sip:30:56: error: ‘PyString_FromString’ was not declared in this scope
         newstring = PyString_FromString(sipCpp->c_str());
                                                        ^
CMakeFiles/python_module_PyKDL.dir/build.make:257: recipe for target 'CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart7.o' failed
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart7.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/python_module_PyKDL.dir/all' failed
make[1]: *** [CMakeFiles/python_module_PyKDL.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'python_orocos_kdl': 
  Command '['/home/aldeb/ros_catkin_ws/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2

Mizux, 2015-08-06:
Fix the pb using:
```
# quick patch
pushd
cd ~/ros_catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL
rm -f ./std_string.sip
wget https://raw.githubusercontent.com/orocos/orocos_kinematics_dynamics/master/python_orocos_kdl/PyKDL/std_string.sip
popd
```

On my way for ROS desktop (not full)...

src: http://answers.ros.org/question/212800/problems-building-from-source/

Mizux, 2015-08-06:
Again error in geneur can't find key-array something -> get the master...
```
# quick patch 2
pushd
cd ~/ros_catkin_ws/src/geneus/src/geneus
rm -f ./geneus_main.py
wget https://raw.githubusercontent.com/jsk-ros-pkg/geneus/master/src/geneus/geneus_main.py
popd
```

Mizux, 2015-08-07:
Jade don't provide urdfdom-header (cf jade-desktop-wet.rosinstall) BUT urdf_parser_plugin need it...
Can be found in aur...
```
yaourt -S urdfdom-headers
```

Edit: just find http://docs.ros.org/independent/api/rosdep/html/contributing_rules.html
try to do the necessary patch soon !

vrabaud, 2015-08-07:
@Mizux , those problems need to be proper pull requests on https://github.com/ros/rosdistro
Please close if that answers all your problems.

Mizux, 2015-08-07:
Indeed, wrong repo...

closed 2015-08-07 by Mizux