ros-infrastructure/rosdep #396: ERROR: Rosdep experienced an internal error.

cameronfabbri, 2015-08-07:
Ran on a Macbook pro with OS X Yosemite 10.10.4

$ rosdep install --from-paths src --ignore-src --rosdistro hydro -y

ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.2

Bad installer [homebrew]: Error while parsing brew info for 'tango-icon-theme'
 * Output of `brew info tango-icon-theme --json=v1`:
 
 * Error while parsing:
 Traceback (most recent call last):
  File "//anaconda/lib/python2.7/site-packages/rosdep2/platforms/osx.py", line 184, in is_installed
    pkg_info = json.loads(std_out)
  File "//anaconda/lib/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "//anaconda/lib/python2.7/json/decoder.py", line 366, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "//anaconda/lib/python2.7/json/decoder.py", line 384, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

wjwwood, 2015-08-07:
@cameronfabbri can you post the output of `brew info tango-icon-theme --json=v1`?

cameronfabbri, 2015-08-07:
I understand the problem, but not sure how to fix it.  Looks like there's multiple targets.  Output:

brew info tango-icon-theme --json=v1
Error: Formulae found in multiple taps: 
 * ros/deps/tango-icon-theme
 * ros/hydro/tango-icon-theme

Please use the fully-qualified name e.g. ros/deps/tango-icon-theme to refer the formula.

wjwwood, 2015-08-07:
There's no good way to have the dependencies for multiple distro tapped at the same time. You can do `brew untap ros/hydro`. The other option is to change the rosdep rule to include the full name:

http://answers.ros.org/question/50660/custom-rosdep-rules/?answer=50681#post-id-50681

closed 2015-09-23 by wjwwood