ros-infrastructure/rosdep #397: Invalid rviz repository (Homebrew tap attempt)

spaghetti-, 2015-08-09:
```
$ rosdep install --from-paths src --ignore-src --rosdistro indigo -y                                                            1 ↵
Error: No available formula for ros/indigo/rviz
Please tap it and then try again: brew tap ros/indigo
executing command [brew install ros/indigo/rviz]
==> Tapping ros/indigo
Cloning into '/usr/local/Library/Taps/ros/homebrew-indigo'...
remote: Repository not found.
fatal: repository 'https://github.com/ros/homebrew-indigo/' not found
```

I have never experienced this myself when I installed Indigo previously.

wjwwood, 2015-08-10:
You're trying to install dependencies for a package which depends on `rviz`, but your `ROS_PACKAGE_PATH`, nor the packages found in `src`, contain `rviz`. So `rosdep` is naively trying to install it from Homebrew, but fails because there is no `rviz` formula in Homebrew. So this is expected behavior, and the work around is to get `rviz` or tell `rosdep` to ignore `rviz` with `--skip-keys`.

closed 2015-08-10 by wjwwood

spaghetti-, 2015-08-11:
I didn't know about `--skip-keys`. I was manually pruning rviz out. Thanks for the heads up.