ros-infrastructure/rosdep #400: Rosdep experienced an error: [Errno 13]

romandy, 2015-08-20:
Ihave a serious problem. When I type the next command in the terminal in Ubuntu:
rosdep install --from-paths src --ignore-src --rosdistro indigo

I get the below message:

ERROR: Rosdep experienced an error: [Errno 13] Permission denied: '/home/romandy/.ros/rosdep/sources.cache/index'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 345, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 422, in _package_args_handler
    lookup = _get_default_RosdepLookup(options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 127, in _get_default_RosdepLookup
    verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 607, in create_default
    sources = load_cached_sources_list(sources_cache_dir=sources_cache_dir, verbose=verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 504, in load_cached_sources_list
    with open(cache_index, 'r') as f:
IOError: [Errno 13] Permission denied: '/home/romandy/.ros/rosdep/sources.cache/index'

I checked the path of Permission denied and I don't find it. I'm not sure it is the error. I hope someone can help me. Yours sincerely, Andres

wjwwood, 2015-08-20:
This is probably because you did `sudo rosdep update` instead of `rosdep update`. You should be able to resolve the issue by doing:

```
$ sudo chmod --reference=~ -R ~/.ros/
```

If this works, please close the issue. Thanks!

romandy, 2015-08-30:
Thanks for the help, fortunately I could resolve the problem add the node 'joint state publisher' in the robot_state_visualize.
Thank you

romandy, 2015-08-30:
But now I haver other issue.

When execute the below line to visualize RVIZ:

roslaunch fanuc_m20ia_support robot_state_visualize_m20ia.launch robot_ip:=172.22.61.233 use_bswap:=false

I get the follow message in the terminal, so, I can´t watch the movement reflected in RVIZ that I do in RoboGuide simulator :

[ERROR] [1440894756.314104745]: Failed to connect to server, rc: -1. Error: 'Connection timed out' (errno: 110)
[ERROR] [1440894756.314598259]: Failed to receive message length
[ERROR] [1440894756.314736100]: Failed to receive incoming message

Thank you for your answers.

Yours sincerely, Andres

tfoote, 2015-08-31:
Glad your rosdep problem is solved. I'm going to close this issue. 

Please ask this as a fully self contained question on http://answers.ros.org with a full description how to reproduce the problem including setting up your environment as well as the commands your running and the full console output.

closed 2015-08-31 by tfoote