ros-infrastructure/rosdep #401: Debian Jessie - rosdep fails to detect libogre-1.8-dev from repos and python-wxgtk2.8 from source

aleksandaratanasov, 2015-08-23:
I'm currently trying to build ROS from source on Jessie using the Wheezy guide. Except for the python-wxgtk2.8 and the python-argparse Jessie seems to have everything covered in the official repositories (of course the mentioned urdf (+headers), collada-dom and console-bridge are to be handled as mentioned in the tutorial; the liblz4-dev is in the repos and detected properly). The required libogre-dev version 1.8 is in the repos too. However **rosdep** fails to detect it. Further I have downloaded the sources fro wxwidgets 2.8 containing python-wxgtk2.8 (from the Debian repos of Wheezy) and compiled those. Using checkinstall I have renamed them to python-wxgtk2.8 but **rosdep** still fails to detect them. I tried locating where exactly this is happening when I call rosdep but couldn't find anything. For python-wxgtk2.8 I need to change the procedure rosdep invokes to do the same as with the URDF, console-bridge and collada-dom and not try to **sudo install** from the repos. For libogre I have no idea why it isn't working. Any ideas how to do that? Haven't given python-argparse a shot yet. Probably will have the same issue.

aleksandaratanasov, 2015-08-23:
Okay, I managed to detect the python-wxgtk2.8. It seemed I have forgotten to add the version number hence the package building failed. Rosdep seems to find it now. The rest still remains as an open issue.

aleksandaratanasov, 2015-08-23:
#368 is already dealing with python-argparse issue. Sorry about including it here.

aleksandaratanasov, 2015-08-23:
Upgrade of libogre-dev to 1.9 doesn't help resolve the issue.

aleksandaratanasov, 2015-08-24:
RViz build fails due to Ogre dependency issue if using 1.9.0. Downgrading to 1.8.0 (see #2805) fixed it and RViz has been built successfully. It seems that a workaround for the issue with **rosdep** not detecting libogre-dev is to build RViz manually (I had to because of the **make -jX** (X != 2) issue also mentioned in the installation guide for Raspberry Pi).

aleksandaratanasov, 2015-08-24:
Build of ROS Indigo on Jessie successful. Can open RViz too :dancer: It seems that the rosdep dependencies are similar to the ones mentioned about the python-rosdep, python-catkin-pkg, python-rospkg and python-rosdistro missing (install for Rasbian for RPi2) - can be totally dismissed.

aleksandaratanasov, 2015-08-24:
I think I know why rosdep fails to detect libogre-dev. The solution is hidden in the terminal output:

    ERROR: the following rosdeps failed to install
      apt: command [sudo -H apt-get install -y libogre-dev] failed
      apt: Failed to detect successful installation of [python-argparse]
      apt: Failed to detect successful installation of [libogre-dev]

In Debian Jessies repos the package is named differently:

    libogre-1.8.0      libogre-1.8-dev    libogre-1.9.0-dbg  libogre-perl
    libogre-1.8.0-dbg  libogre-1.9.0      libogre-1.9-dev

Unlike in Wheezy: https://packages.debian.org/wheezy/libogre-dev

I presume this has been changed in ROS Jade so there is no need of opening a new issue and a pull request for this.

wjwwood, 2015-09-01:
You shouldn't need any `wx` packages. What version of ROS are you building?

If the key for ogre resolves to the wrong name on Jessie, then you need to update the rosdep rules:

https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L1675

It looks like you've figure out the problems? I'll close this. Please continue to comment if there are more `rosdep` specific issues.

closed 2015-09-01 by wjwwood

aleksandaratanasov, 2015-09-01:
During last rebuild I didn't got the libogre issue (however it is possible that I have missed it due to the huge amount of output). Will be more careful next time I rebuild after adding some package and if the issue is still there we can reopen this one here. As for the version - it is ROS Indigo. I was also surprised since - if I remember correctly - `wx` packages were a must when ROS actually was using wxWidgets (rxgraph, rxbag etc.). These are all depricated now and I haven't downloaded any ROS package that requires wxWidgets (I'm pretty much manually checking every single ROS package I install for dependency issues).

wjwwood, 2015-09-01:
Ok, if it persists, comment again, and I'll reopen.

tfoote, 2015-09-01:
As of groovy we switched from wx to qt: [groovy release notes](http://www.ros.org/news/2012/12/ros-groovy-galapagos-released.html)

davidrotor, 2016-01-04:
aleksan could you tell me how to checkinstall the python-pwxgtk2.8 package from source i did it but rosdep doesnt find it

davidrotor, 2016-01-04:
if I use OS:Debian:jessie I always get ImportError: "from catkin_pkg.topological_order import topological_order" failed: No module named tool_detection

NikolausDemmel, 2016-01-05:
> if I use OS:Debian:jessie I always get ImportError: "from catkin_pkg.topological_order import topological_order" failed: No module named tool_detection

What version of `rosdep` and `catkin_pkg` do you use? Please make sure both are up-to-date.