ros-infrastructure/rosdep #402: rosdep tries to install vtk-phython

sorki, 2015-08-26:
Typo - it should try to install ``vtk-python`` but I'm not able to find correct place to fix it.

tfoote, 2015-08-26:
Please be a little bit more specific about your problem. It's important to provide enough information to be able to reproduce your problem. What platform are you running on? What command are you running? What rosdep sources are you using? 

From digging around I did find a rule which matches your misspelling. [here](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L2886)

There is a correct rule [here](https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L1951) and it doesn't have your misspelling.

The incorrect rule should be trimmed and the gentoo rule should be moved into python.yaml and the whole key purged from base.yaml. 

Could you provide a PR for that?

cottsay, 2015-08-26:
My fault :( sorry...

sorki, 2015-09-22:
Seems to be fixed, thanks!

closed 2015-09-22 by sorki