ros-infrastructure/rosdep #406: Homebrew/homebrew Error: Operation already in progress for protobuf

mzaiady, 2015-09-02:
On Mac OSX:10.10.5

$rosdep install --from-paths src --ignore-src --rosdistro indigo -y

executing command [brew install gazebo]
==> Installing gazebo from osrf/homebrew-simulation
Error: Operation already in progress for protobuf
Another active Homebrew process is already using protobuf.
Please wait for it to finish or terminate it to continue.
ERROR: the following rosdeps failed to install
homebrew: command [brew install gazebo] failed

wjwwood, 2015-09-02:
This isn't a rosdep error, you can see from this:

```
Error: Operation already in progress for protobuf
Another active Homebrew process is already using protobuf.
Please wait for it to finish or terminate it to continue.
```

That you are running Homebrew in parallel and it is having resource contention.

mzaiady, 2015-09-02:
No other Homebrew running, i did restart the machine , but still the same result!

a comment in  Homebrew suggests that it this could happen is a tap has a formula that duplicates the name of a formula in core. We can provide some assistance if there are questions.

wjwwood, 2015-09-02:
I'll reopen the issue until this is resolved, but I do not believe this is an issue with rosdep.

wjwwood, 2015-09-02:
@mzaiady you can work around this by installing `gazebo2` from homebrew and skipping this key with rosdep:

```
$ brew install gazebo2
$ rosdep install --from-paths src --ignore-src --rosdistro indigo -y --skip-keys gazebo
```

mzaiady, 2015-09-03:
Executing the above, a new error show up:
 
$ rosdep install --from-paths src --ignore-src --rosdistro indigo -y --skip-keys gazebo

executing command [brew install graphviz]
Warning: graphviz-2.38.0 already installed, it's just not linked
ERROR: the following rosdeps failed to install
  homebrew: Failed to detect successful installation of [graphviz]

Note: I'm new to both homebrew and ROS, so may be there is a simple way to fix this, but I'm not aware of.

wjwwood, 2015-09-04:
`brew link graphviz`?

mzaiady, 2015-09-05:
$ brew link graphviz
Linking /usr/local/Cellar/graphviz/2.38.0... 
Error: Could not symlink bin/dot
Target /usr/local/bin/dot
already exists. You may want to remove it:
  rm '/usr/local/bin/dot'

-- removing   '/usr/local/bin/dot' and tried to link again , I got 

Error: Could not symlink bin/dot2gxl
Target /usr/local/bin/dot2gxl
already exists. You may want to remove it:
  rm '/usr/local/bin/dot2gxl'

and so on,

Error: Could not symlink bin/gc
Target /usr/local/bin/gc
already exists. You may want to remove it:
  rm '/usr/local/bin/gc'

Should I keep deleting? I think this may break something else!

I tried  to jump to next step by running 
$ ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release

Some other issues and missing dependancies (I think because of previous step did not complete)

I solved some like QT version issue, my machine had 4.8.7  but it look like the code requires 4.8.6, I found some workarounds to overcome this, but  now I'm getting

make -j8 -l8 in '/Users/mohamedalzaiady/ros_catkin_ws/build_isolated/geometric_shapes'
[ 14%] Building CXX object CMakeFiles/geometric_shapes.dir/src/mesh_operations.cpp.o
/Users/mohamedalzaiady/ros_catkin_ws/src/geometric_shapes/src/mesh_operations.cpp:54:10: fatal error: 'assimp/aiScene.h' file not found
#include <assimp/aiScene.h>
         ^
1 error generated.
make[2]: *** [CMakeFiles/geometric_shapes.dir/src/mesh_operations.cpp.o] Error 1
make[1]: *** [CMakeFiles/geometric_shapes.dir/all] Error 2
make: *** [all] Error 2

I did install assimp, but it did not slove it

I came from java development , so I'm not familiar of such issues or building from source, and I'm trying to do my best searching online and trying all options/workarounds before posting the issue, but because of  lack of knowledge (for now) about how such build systems work and how to adjust it to my needs, I'm not able to proceed by my own. 

Any ideas how to over come graphviz link and how to continue?

mzaiady, 2015-09-05:
Also running brew doctor gives a lot of warnings, i did try to solve as much as I can and searched online to find solutions,  but I couldn't solve much, do you want  me to post the 'brew doctor' result?

wjwwood, 2015-09-10:
> Should I keep deleting? I think this may break something else!

This sounds like you have previously installed graphviz outside of Homebrew. I guess you can continue to replace things. You can also use the `--overwrite` option with `brew link` to have it remove them for you.

> I did install assimp, but it did not solve it.

I don't know about that, I was unable to reproduce this and I have assimp installed:

```
% brew info assimp
assimp: stable 3.1.1 (bottled), HEAD
Portable library for importing many well-known 3D model formats
http://assimp.sourceforge.net/
/usr/local/Cellar/assimp/3.1.1 (53 files, 8.3M) *
  Poured from bottle
From: https://github.com/Homebrew/homebrew/blob/master/Library/Formula/assimp.rb
[ ... ]
```

> Also running brew doctor gives a lot of warnings

Well it sounds like a lot of your issues are coming from your system having existing problems. Perhaps the easiest solution would be to start with a Ubuntu VM until you get a better handle on dependency management and compilation with C++ software.

mzaiady, 2015-09-13:
Installation on Ubuntu VM done, without no issues at all :)

For the OSX,  I Cleaned up all 'brew doctor' issues, by uninstall mac ports and python 3,  except one issue (I think it's not related),

Now, running 
 rosdep install --from-paths src --ignore-src --rosdistro indigo -y

Produces :
#All required rosdeps installed successfully

But, still running 
$ ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release

Gives 

==> Processing catkin package: 'qt_gui_cpp'
==> Building with env: '/Users/mohamedalzaiady/ros_catkin_ws/install_isolated/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/Users/mohamedalzaiady/ros_catkin_ws/build_isolated/qt_gui_cpp'
==> make -j8 -l8 in '/Users/mohamedalzaiady/ros_catkin_ws/build_isolated/qt_gui_cpp'
[ 78%] Built target qt_gui_cpp
[ 85%] Compiling generated code for qt_gui_cpp_sip Python bindings...
ld: warning: directory not found for option '-L/usr/local/Cellar/qt/4.8.6/lib'
ld: warning: directory not found for option '-F/usr/local/Cellar/qt/4.8.6/lib'
ld: framework not found QtCore
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** ["/Users/mohamedalzaiady/ros_catkin_ws/devel_isolated/qt_gui_cpp/lib/python2.7/site-packages/qt_gui_cpp/libqt_gui_cpp_sip".so] Error 1
make[2]: *** [/Users/mohamedalzaiady/ros_catkin_ws/devel_isolated/qt_gui_cpp/lib/python2.7/site-packages/qt_gui_cpp/libqt_gui_cpp_sip.dylib] Error 2
make[1]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'qt_gui_cpp': 

I noticed that brew installed qt version 4.8.7, while it look like the build  searching for 4.8.6 !

wjwwood, 2015-09-14:
@mzaiady Please search before asking and from now on, please post on answers.ros.org, since these are not rosdep issues. I believe that what you've described is a known issue:

http://answers.ros.org/question/211546/fatal-error-qmetatype-file-not-found-when-building-ros-indigo-on-os-x-10103/?answer=212209#post-id-212209

wjwwood, 2015-09-14:
Also I closed this because you've cleared up the installation of dependencies with `rosdep` by cleaning up your system and the protobuf/gazebo issue and the Homebrew error not being descriptive are both ticked upstream.

closed 2015-09-14 by wjwwood

mzaiady, 2015-09-16:
Thank you for your support :+1: