ros-infrastructure/rosdep #414: unable to resolve dep Bad installer [pip]:.......

Lome789, 2015-09-25:
When i input       rosdep install --from-paths src --ignore-src --rosdistro jade   
it shows
 ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.3

Bad installer [pip]: Command '['pip', 'show', 'nose']' returned non-zero exit status 1

wjwwood, 2015-09-25:
This may be related to https://github.com/ros-infrastructure/rosdep/pull/413

wjwwood, 2015-09-25:
I can reproduce this, I have proposed https://github.com/ros-infrastructure/rosdep/pull/415 as a fix.

Lome789, 2015-09-25:
Sorry  .... Actually i have no idea about how to fix it with ur method.. i am a complete newbie..

wjwwood, 2015-09-25:
@Lome789 that's ok, I'm going to make a new release asap and then you can just update rosdep and try again. I'm just making sure the fix is good.

closed 2015-09-25 by wjwwood

Lome789, 2015-09-25:
Thanks a lot!

wjwwood, 2015-09-25:
Ok, `0.11.4` should be out now, so you can update `rosdep` and try what ever you were doing again.

You can check your `rosdep` version like this:

```
% rosdep --version
0.11.4
```

wkentaro, 2015-09-26:
@wjwwood
Thanks. That works in my environment.

Lome789, 2015-09-26:
No matter how many times i updated rosdep it was still 0.11.3 
i have ran sudo rosdep update once maybe it would cause problem?

wkentaro, 2015-09-26:
I think I don't understand what you mean with sudo rosdep update, but the problem of this issue will be resolved after you upgrade rosdep to 0.11.4.

Lome789, 2015-09-26:
yeah ...unfortunately i can't upgrade to 0.11.4

wkentaro, 2015-09-26:
rosdep for apt is not yet released.
maybe you should wait for it, or remove apt package and install via pip.

Lome789, 2015-09-26:
ok..thank u so much!

tfoote, 2015-09-26:
0.11.4 is available via debian package

Lome789, 2015-09-26:
In OS X how to achieve this update?

wkentaro, 2015-09-26:
```
$ pip install --upgrade rosdep
$ rosdep --version
```

Lome789, 2015-09-26:
ew...i gonna try to find other related issues... 

$ pip install --upgrade rosdep
Collecting rosdep
  Using cached rosdep-0.11.4.tar.gz
Requirement already up-to-date: catkin-pkg in /Library/Python/2.7/site-packages (from rosdep)
Requirement already up-to-date: rospkg>=1.0.34 in /Library/Python/2.7/site-packages (from rosdep)
Requirement already up-to-date: rosdistro>=0.4.0 in /Library/Python/2.7/site-packages (from rosdep)
Requirement already up-to-date: PyYAML>=3.1 in /Library/Python/2.7/site-packages (from rosdep)
Requirement already up-to-date: argparse in /Library/Python/2.7/site-packages (from catkin-pkg->rosdep)
Requirement already up-to-date: docutils in /Library/Python/2.7/site-packages (from catkin-pkg->rosdep)
Requirement already up-to-date: python-dateutil in /Library/Python/2.7/site-packages (from catkin-pkg->rosdep)
Requirement already up-to-date: setuptools in /Library/Python/2.7/site-packages (from rosdistro>=0.4.0->rosdep)
Requirement already up-to-date: six>=1.5 in /Library/Python/2.7/site-packages (from python-dateutil->catkin-pkg->rosdep)
Installing collected packages: rosdep
  Found existing installation: rosdep 0.11.3
    Uninstalling rosdep-0.11.3:
Exception:
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/basecommand.py", line 211, in main
    status = self.run(options, args)
  File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/commands/install.py", line 311, in run
    root=options.root_path,
  File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/req/req_set.py", line 640, in install
    requirement.uninstall(auto_confirm=True)
  File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/req/req_install.py", line 716, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/req/req_uninstall.py", line 125, in remove
    renames(path, new_path)
  File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/utils/__init__.py", line 315, in renames
    shutil.move(old, new)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 300, in move
    rmtree(src)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 252, in rmtree
    onerror(os.remove, fullname, sys.exc_info())
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 250, in rmtree
    os.remove(fullname)
OSError: [Errno 13] Permission denied: '/Library/Python/2.7/site-packages/rosdep-0.11.3-py2.7.egg-info/dependency_links.txt'

wkentaro, 2015-09-26:
sudo ?

Lome789, 2015-09-26:
yeah it works!!!thanks a lot!!!

PR #415 (2015-09-25): use popen to check output and return code of 'pip show ...'
Should fix #414.