ros-infrastructure/rosdep #416: ERROR: Rosdep experienced an internal error. (Bad Installer [pip] ...)

japzi, 2015-09-26:
OSX Yosamite 10.10.3
$ rosdep install --from-paths src --ignore-src --rosdistro jade -y --skip-keys ogre --skip-keys gazebo
ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.3

Bad installer [pip]: Command '['pip', 'show', 'nose']' returned non-zero exit status 1

wjwwood, 2015-09-26:
I believe this is a duplicate of https://github.com/ros-infrastructure/rosdep/issues/414

closed 2015-09-26 by wjwwood

japzi, 2015-09-26:
yes it is, didn't realize a new version 0.11.4 was release. went through the step without an issue with above 0.11.4.

wjwwood, 2015-09-26:
@japzi no problem, sorry for the interruption.