ros-infrastructure/rosdep #417: rosdep fails because of too old pip (installed via apt on ubuntu 12.04)

wkentaro, 2015-09-26:
bacause `pip show` does not exists for pip version 1.0. (the released version of python-pip on ubuntu 12.04)
package release of python-pip: http://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=python-pip&searchon=names 

but rosdep requires `pip show` (see: https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/platforms/pip.py#L81-L84)

I don't know how to resolve this for maintainers of rosdep.
I'm using latest pip (installed via easy_install and upgrade with pip install -U) as user.

maybe we need warning for too old pip? @wjwwood

wkentaro, 2015-09-26:
I see hydro (on Ubuntu 12.04) is deprecated and personally using indigo on Ubuntu 14.04,
but some my lab members still use hydro and I think that is same for other users of ROS.

wjwwood, 2015-09-29:
@wkentaro What version of `rosdep` is this an issue for? Did my latest release, `0.11.4`, fix this?

wkentaro, 2015-09-29:
Actually, no. 
The pip version on Ubuntu12.04 is 1.0. 
And it has no show subcommand.

wjwwood, 2015-09-29:
@wkentaro But, it should be in `pip freeze` right? What is the actual error. Does `rosdep` fail or does it fail to resolve that the packages are install?

wkentaro, 2015-09-29:
I encountered this problem at https://github.com/jsk-ros-pkg/jsk_demos/pull/1065 in travis test for hydro.
(Actually, I don't use hydro so cannot reproduce in my local PC)

The error output is:

```
+ rosdep install -r -q -n --from-paths . /opt/ros/hydro/share --ignore-src --rosdistro hydro -y
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
Usage: pip COMMAND [OPTIONS]

pip: error: No command by the name pip show
  (maybe you meant "pip install show")

ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.3

Bad installer [pip]: Command '['pip', 'show', 'fysom']' returned non-zero exit status 2
```

This seems failed because of no `pip show` command, so even if 0.11.4, it should fails.

Maybe we can skip this block if pip version is old.
https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/platforms/pip.py#L78-L90

wjwwood, 2015-10-12:
I don't think the above error is using the code from master which you linked to, because the linked to code uses `subprocess.Popen` which is does not raise a `CalledProcessError`, which is what is being caught and reported in the output above.

wkentaro, 2015-10-12:
the log says rosdep version is 0.11.3.
so correctly here:
https://github.com/ros-infrastructure/rosdep/blob/e7c0053cdefe2f4fbaa553f2294be9f17e8d586f/src/rosdep2/platforms/pip.py#L78-L84

wjwwood, 2015-10-12:
Then that's the issue, because `subprocess.checkout_output` raises a `CalledProcessError` if the return code is not `0`. The above behavior should not happen with the code on master.

wkentaro, 2015-10-13:
I see. I tested with `rosdep==0.11.4` and `pip==1.0.0`. I had no error at that time, so I close this.

closed 2015-10-13 by wkentaro