ros-infrastructure/rosdep #420: rosdep successfully installs pip package, but then has error: failed to detect pip package

peterpolidoro, 2015-09-28:
rosdep used to work with my setup and now it fails with a strange error and I am not sure what changed. I have a Trusty/Jade installation using python 2.7 and pip 1.5.4. Perhaps the problem is not with rosdep, but I am not sure where else to look. One of my ros packages depends on a pip package. In my catkin workspace I run "rosdep install --from-paths ." I get the following error:

    Successfully installed my-pip-package
    Cleaning up...
    ERROR: the following rosdeps failed to install
        pip: Failed to detect successful installation of [my_pip_package]

If I run "rosdep install --from-paths ." again I get the following error:

    Requirement already up-to-date: my-pip-package in /usr/local/lib/python2.7/dist-packages
    ERROR: the following rosdeps failed to install
        pip: Failed to detect successful installation of [my_pip_package]

wjwwood, 2015-09-28:
@peterpolidoro This might be related to #412/#415? I'll try to look into it as soon as I get a chance. A workaround might be to downgrade your version of `rosdep` to `0.11.2` or older, but that might have other issues.

wjwwood, 2015-09-30:
@peterpolidoro Can you post the output of `pip show my_pip_package`? As well as check the return code (`echo $?`)?

peterpolidoro, 2015-09-30:
No output from `pip show my_pip_package`. The return code is 0. If I run `pip install my_pip_package`, the output is `Requirement already satisfied`. `pip -V` outputs `pip 1.5.4 from /usr/lib/python2.7/dist-packages (python 2.7)`.

wjwwood, 2015-09-30:
Can you verify what version of `rosdep` you're using with `rosdep --version`?

peterpolidoro, 2015-09-30:
0.11.4

wjwwood, 2017-07-28:
I was just catching up on this issue again. I was still unable to reproduce what you're talking about. While working on another issue (different project) about a year ago, I do remember some changes in pip related to this.

@peterpolidoro, or anyone else really, do you know if this is still a problem?

I'm going to close for now, but I can reopen if it is.

closed 2017-07-28 by wjwwood

alecive, 2018-02-10:
@wjwwood I am currently having this issue on my docker builds : https://hub.docker.com/r/scazlab/human_robot_collaboration/builds/bjv5btpo5fencqmzdclekv6/

Relevant code snippet (at the end of the page I linked above):
 
```
Downloading/unpacking backports.ssl-match-hostname

  Downloading backports.ssl_match_hostname-3.5.0.1.tar.gz

  Running setup.py (path:/tmp/pip_build_root/backports.ssl-match-hostname/setup.py) egg_info for package backports.ssl-match-hostname

    

Installing collected packages: backports.ssl-match-hostname

  Running setup.py install for backports.ssl-match-hostname

    

Successfully installed backports.ssl-match-hostname
Cleaning up...

[91mERROR: the following rosdeps failed to install
  pip: Failed to detect successful installation of [backports.ssl_match_hostname]
[0m
[1mexecuting command [apt-get install -y libnlopt-dev][0m
[1mexecuting command [apt-get install -y ros-indigo-rospy-tutorials][0m
[1mexecuting command [apt-get install -y python-bson][0m
[1mexecuting command [apt-get install -y ros-indigo-pr2-description][0m
[1mexecuting command [apt-get install -y ros-indigo-rosauth][0m
[1mexecuting command [apt-get install -y python-twisted-core][0m
[1mexecuting command [apt-get install -y ros-indigo-moveit-core][0m
[1mexecuting command [apt-get install -y ros-indigo-tf-conversions][0m
[1mexecuting command [apt-get install -y libnlopt0][0m
[1mexecuting command [apt-get install -y ros-indigo-urdf][0m
[1mexecuting command [pip install -U backports.ssl_match_hostname][0m

Removing intermediate container 3fb02a9ca367

The command '/bin/bash -c cd ~/ros_ws      && rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO' returned a non-zero code: 1
click
```

wjwwood, 2018-02-13:
@alecive Hmm, I'm not sure. I don't really have time to dig into the issue, but if you come up with a root cause maybe I can help get a fix in. Just glancing at your `Dockerfile` maybe you could try jumping into it at the end and running the rosdep command again but not as the root user.

alecive, 2018-02-15:
Although running `rosdep` without root might solve the issue, I don't think it is advisable. Furthermore:

 * This issue appeared only in the past two weeks, whereas it was working fine in the past year or so
 * I have another `Dockerfile` that does the same thing without issues ( [this one](https://hub.docker.com/r/scazlab/hrc-docker/) )

Also, if I install the package before running rosdep, i.e. by adding the following:

```
RUN pip2 install backports.ssl-match-hostname
RUN pip2 show backports.ssl-match-hostname
```

The error becomes:
```
Requirement already up-to-date: backports.ssl-match-hostname in /usr/local/lib/python2.7/dist-packages
Cleaning up...
ERROR: the following rosdeps failed to install
  pip: Failed to detect successful installation of [backports.ssl_match_hostname]
```

alecive, 2018-02-15:
@wjwwood the answer apparently is that pip changed. I fixed my issue by following this github issue https://github.com/start-jsk/jsk_apc/issues/1527 and running `pip install pip --upgrade` as root in my `Dockerfile` before running `rosdep`.

ipa-mdl, 2018-06-05:
@alecive: pip replaces dots with dashes internally, so ` pip2 install backports-ssl-match-hostname` should work without upgrading pip.

martino92, 2022-04-12:
> 

It couldn't solve the issue in my case...