ros-infrastructure/rosdep #421: rosdep update failed with EI Capitan OS X

Rangozhang, 2015-10-03:
Everything works well until EI Capitan.

When I run "rosdep update", I get this:

reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"

ERROR: Rosdep experienced an error: unrecognized version: 10.11
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.4

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 545, in command_update
    error_handler=update_error_handler)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/sources_list.py", line 466, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/gbpdistro_support.py", line 147, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/local/lib/python2.7/site-packages/rosdep2/__init__.py", line 85, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/platforms/debian.py", line 50, in register_platforms
    register_linaro(context)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/platforms/debian.py", line 63, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/local/lib/python2.7/site-packages/rosdep2/installers.py", line 112, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/local/lib/python2.7/site-packages/rospkg/os_detect.py", line 518, in get_name
    self.detect_os()
  File "/usr/local/lib/python2.7/site-packages/rospkg/os_detect.py", line 481, in detect_os
    self._os_codename = os_detector.get_codename()
  File "/usr/local/lib/python2.7/site-packages/rospkg/os_detect.py", line 253, in get_codename
    return _osx_codename(*ver[0:2])
  File "/usr/local/lib/python2.7/site-packages/rospkg/os_detect.py", line 232, in _osx_codename
    raise OsNotDetected("unrecognized version: %s.%s"%(major, minor))
OsNotDetected: unrecognized version: 10.11

How can we get around it? Thanks!

tfoote, 2015-10-03:
This PR should fix your issue: https://github.com/ros-infrastructure/rospkg/pull/87

eduran0826, 2015-10-13:
I am having the same issue.

tfoote, 2015-10-13:
@eduran0826  What version of rospkg are you using? It should have been fixed in 1.0.36

tfoote, 2015-10-30:
Resolved in rospkg.

closed 2015-10-30 by tfoote