ros-infrastructure/rosdep #422: Fedora 23 fail

MalcolmMielle, 2015-10-06:
When doing rosdep on Fedora 23, I have this : 

```
rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"

ERROR: Rosdep experienced an error: 'release' is not in list
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/site-packages/rosdep2/main.py", line 343, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/site-packages/rosdep2/main.py", line 351, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/site-packages/rosdep2/main.py", line 533, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/site-packages/rosdep2/sources_list.py", line 466, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/site-packages/rosdep2/gbpdistro_support.py", line 147, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python2.7/site-packages/rosdep2/__init__.py", line 85, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python2.7/site-packages/rosdep2/platforms/debian.py", line 50, in register_platforms
    register_linaro(context)
  File "/usr/lib/python2.7/site-packages/rosdep2/platforms/debian.py", line 63, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python2.7/site-packages/rosdep2/installers.py", line 112, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python2.7/site-packages/rospkg/os_detect.py", line 518, in get_name
    self.detect_os()
  File "/usr/lib/python2.7/site-packages/rospkg/os_detect.py", line 480, in detect_os
    self._os_version = os_detector.get_version()
  File "/usr/lib/python2.7/site-packages/rospkg/os_detect.py", line 176, in get_version
    idx = os_list.index('release')
ValueError: 'release' is not in list
```

Thanks,

wjwwood, 2015-10-06:
I guess it's because something needs to be changed for Fedora 23 to be supported, but I don't know. Either way the error handling could probably be better. @cottsay any insight?

MalcolmMielle, 2015-10-06:
Maybe I should move this issue to rospkg since the problem seem to be on the OS detection. What do you think ?

Anyway, I've been playing around with rospkg and this is what I got when printing os_list in os_detect.py : 
 
```
[u'\\S', u'Kernel', u'\\r', u'on', u'an', u'\\m', u'(\\l)']
```

Just for fun I changed 'release' to '\\r' and then I have an output : 

```
OS Name:     fedora
OS Version:  on
OS Codename: twenty
```

I don't know why there is no release in the list...

For information 
```
more /etc/fedora-release 
Fedora release 23 (Twenty Three)
```

cottsay, 2015-10-06:
@MalcolmMielle - Assuming you're using the newest rosdep RPM for f23 (`0.11.2-4.fc23`), I think this is probably because it is in beta.

Fedora version detection should be using `/etc/redhat-release`. Please post the contents of that file. My bet is that "Beta" or "pre-release" is somewhere in there, and it mucks with rospkg's parsing.

--scott

MalcolmMielle, 2015-10-06:
This is indeed the version of rosdep I'm using 0.11.2-4.fc23.

The content of `redhat-release` is 

```
Fedora release 23 (Twenty Three)
```

Sadly, no beta or pre-release.

I think the problem might come from the file `os_detect.py` in rospkg. In the class `Fedora(OsDetector)` : 

```
class Fedora(OsDetector):
    """
    Detect Fedora OS.
    """
    def __init__(self, release_file="/etc/redhat-release", issue_file="/etc/issue"):
        self._release_file = release_file
        self._issue_file = issue_file
        
    def is_os(self):
        os_list = read_issue(self._release_file)
        return os_list and os_list[0] == "Fedora" 

    def get_version(self):
        if self.is_os():
            os_list = read_issue(self._issue_file) #<- THIS LINE
            idx = os_list.index('release')
            if idx > 0:
                return os_list[idx+1]
        raise OsNotDetected('cannot get version on this OS')

    def get_codename(self):
        if self.is_os():
            os_list = read_issue(self._release_file)
            idx = os_list.index('release')
            matches = [x for x in os_list if x[0] == '(']
            codename = matches[0][1:]
            if codename[-1] == ')':
                codename = codename[:-1]
            return codename.lower()
        raise OsNotDetected('called in incorrect OS')
```

Shouldn't the line with comment be something like  : `os_list = read_issue(self._release_file)` ? If I get it correctly, for now it's reading `/etc/issue` and not `/etc/redhat-release`.

cottsay, 2015-10-06:
It looks like it is reading from `/etc/issue`, and has been for quite a while.

On Fedora 21, `/etc/issue` looks like:
```
Fedora release 21 (Twenty One)
Kernel \r on an \m (\l)
```

On Fedora 22, `/etc/issue` looks like:
```
Fedora release 22 (Twenty One)
Kernel \r on an \m (\l)
```

So if `/etc/issue` looks like:
```
\S
Kernel \r on an \m (\l)
```
...on f23 like you said it does, then that is possibly the culprit. Try making it look like the others and see if that fixes your problem. If it does, we need to investigate if this is going to be in the final release.

MalcolmMielle, 2015-10-06:
I changed my `\etc\issue` to
```
Fedora release 23 (Twenty Three)
Kernel \r on an \m (\l)
```

And it went through the same way as when I replaced `issue` by `redhat-release` :).

According to [this link](http://askubuntu.com/questions/335695/why-does-etc-issue-show-me-ubuntu-and-not-lubuntu) `\S` inserts the operating system name, so I guess it makes sense. I don't know if that will be in the final release though.

cottsay, 2015-10-06:
Okay, it looks like this is the new behavior for this file.

See [RHBZ#1239089](https://bugzilla.redhat.com/show_bug.cgi?id=1239089), and the subsequent change to the `fedora-release` RPM [1]

We will probably need to update `rospkg` to resolve "\S" to the OS version string, or we need to discuss the ramifications of changing to `/etc/redhat-release` (ie, bloom generation FOR Fedora FROM Ubuntu, etc).

Fedora 23 freeze is on the 13th, and the final release is on the 27th. I would say that this is a blocker for using ROS on Fedora 23.

Thanks,

--scott

FYI @jpgr87

[1] http://pkgs.fedoraproject.org/cgit/fedora-release.git/commit/?id=294e7881

tfoote, 2015-10-06:
This should probably move to a rospkg ticket. If issue is providing that substitution, where is it getting the version?

MalcolmMielle, 2015-10-06:
No worries, if you need more test on Fed23 and I can help, don't hesitate.

There is a `\etc\fedora-release` file with : 
```
Fedora release 23 (Twenty Three)
```
inside as well. I feel like this is pretty much what you're looking for with `\etc\issue` :).

Do you reckon I should fill a bug in rospkg ?

jpgr87, 2015-10-06:
I think this bug is actually in rospkg, not rosdep.  That being said, I wrote a detector that uses /etc/os-release for rospkg that I never got around to pushing upstream.  It works on my f22 machine.

@MalcolmMielle, would you mind posting the contents of your /etc/os-release so I can verify it's compatible with f23 as well?

MalcolmMielle, 2015-10-06:
The os-release file : 

```
NAME=Fedora
VERSION="23 (Workstation Edition)"
ID=fedora
VERSION_ID=23
PRETTY_NAME="Fedora 23 (Workstation Edition)"
ANSI_COLOR="0;34"
CPE_NAME="cpe:/o:fedoraproject:fedora:23"
HOME_URL="https://fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=23
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=23
PRIVACY_POLICY_URL=https://fedoraproject.org/wiki/Legal:P$
VARIANT="Workstation Edition"
VARIANT_ID=workstation
```

jpgr87, 2015-10-06:
Ok my version works, but it's pulling out "Workstation Edition" as the codename.  Given fedora doesn't have codenames anymore, it's probably not the worst result in the world.

I'll open a pull request against rospkg with an updated os detector.

jpgr87, 2015-10-06:
I've posted the pull request as ros-infrastructure/rospkg#88, I think it'll fix this issue

wjwwood, 2015-12-08:
Closing in favor of https://github.com/ros-infrastructure/rospkg/pull/88

closed 2015-12-08 by wjwwood