ros-infrastructure/rosdep #424: rosdep does not support running in non-elevated mode

spaghetti-, 2015-10-14:
https://github.com/ros-infrastructure/rosdep/blob/6cdef863382f5c812030ee1c15e43c4f0b98149e/src/rosdep2/installers.py#L320

The option above is immutable.

I came across this because with system integrity protection in el-capitan, I've moved to using Homebrew Python. I don't need root access for homebrew python to be managed.

If this change is supported I'll submit a PR.

NikolausDemmel, 2015-10-15:
Does the `--as-root` command line option not help in your use case?

tfoote, 2015-10-30:
I believe that this is resolved by `--as-root` option. It's not true that the option is immutable. See the diff for #307 for the implementation. 

If that doesn't resolve your issue please, comment and we can reopen.

closed 2015-10-30 by tfoote