ros-infrastructure/rosdep #426: update default sources list url to use githubusercontent.com

tfoote, 2015-10-27:
It's currently using raw.github.com which is now failing in some cases. 

Refiled from: https://github.com/ros/rosdistro/issues/9721

It's the right thing to do, but I cannot reproduce the error using rosdep init.

wjwwood, 2015-10-27:
I think this line needs to be update: https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/sources_list.py#L69

ostueker, 2015-10-28:
It seems changing the URL like I have suggested does not solve my problems.

I've just tried changing the URL in sources_list.py to:
* https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
* http://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
* http://raw.github.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list

and in all cases I get "Website may be down" from rosdep init.
Also trying to fetch these URLS with wget or curl doesn't work on these (admittedly old) RHEL 4 machine.

Can that have something to do that I don't have (want to use) root privileges for the rosdep init?
I have installed ros in my home directory with the following commands:

```
module load gcc python/2.7.10  cmake
pip install --user --upgrade pip
pip install --user -U rosdep rosinstall_generator wstool rosinstall
python .local/bin/rosdep init
```

What else besides downloading `20-default.list` to `/etc/ros/rosdep/sources.list.d/` does `rosdep init` do? 
Is there a fallback (local) location for the `/etc/ros` tree for a locally contained installation?

We don't want to clutter up our /etc/ on the cluster with configurations of packages that our users have installed in their home directory.

Thanks,
Oliver

tfoote, 2015-10-28:
rosdep init just fetches the file and drops in in the location. It uses urllib under the hood. But your problem is not dropping the file, but fetching the file. If curl and wget can't retrieve the file you'll need to resolve that issue. It suggests your requests are being blocked or are incompatible with something github is doing. Maybe it's using a more modern redirect or something that RHEL4 doesn't support? You will need to dig deeper into the outputs of your fetch attempts. Don't worry about rosdep until you can fetch the files using curl or wget. 

You can change the location rosdep looks for the source files using `ROSDEP_SOURCE_PATH` in your environment to avoid using etc. We don't recommend using it because rosdep database consistency across users is important.

closed 2015-10-29 by wjwwood

artechmalayali, 2023-05-20:
Can you tell me what is the exact solution for the problem confused with all things what to do to solve this?

PR #427 (2015-10-29): change download url for init.
Fixes #426 
Using the new githubusercontent.com url.
commit 2911dc72 (2015-10-29): change download url for init.
Fixes #426