ros-infrastructure/rosdep #430: rosdep install fails for metapackages

ipa-fmw, 2015-11-24:
rosdep install fails to install dependencies from metapackages. 

We tracked down the error to `rospkg_loader.py`:
`get_rosdeps` (https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/rospkg_loader.py#L120) returns an empty list for metapackages. Comment in https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/rospkg_loader.py#L136 is `# stacks currently do not have rosdeps of their own, implicit or otherwise`.

Is this a left-over from pre-catkin and a bug? or is this done by purpose?

Example is the care-o-bot metapackages in https://github.com/ipa320/care-o-bot. Cloning that into an empty catkin workspace and running `rosdep install --from-path src -i -y` does not install any dependencies.

wjwwood, 2017-07-28:
I know this is an old bug, but I'm not sure if this is by design or not. @dirk-thomas do you know of any conceptual reason that metapackages should not get their dependencies installed?

NikolausDemmel, 2017-07-28:
I personally can't think of any reason and agree that it should be changed.

ipa-mdl, 2017-07-28:
Needs to be fixed in [rospkg_loader.py](https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/rospkg_loader.py#L136)

dirk-thomas, 2017-07-28:
I don't see a reason why metapackages should be treated differently either.

PR #531 (2017-07-31): resolve dependencies of meta-packages
fixes #430 

Haven't tested it yet.
For the sake of simplicity meta-package's `build_depends` and `test_depends` get returned as well, even if these should be empty anyway.
closed 2017-07-31 by wjwwood