ros-infrastructure/rosdep #433: missing -R argument

dustingooding, 2016-01-22:
rosdep's documentation lists -R as an install argument, that recursively installs all dependencies for a given package, not just that package's dependencies.

http://docs.ros.org/independent/api/rosdep/html/commands.html

This argument is not supported in the code though.  Is it better to remove -R from the docs, or add -R to the code?

wjwwood, 2016-02-14:
Sorry to leave this hanging so long. I think it would be better to either remove all CLI documentation from the generated docs or have them generated based on the output of `rosdep -h` automatically. I'll look into how to do this.

wjwwood, 2016-02-14:
I created https://github.com/ros-infrastructure/rosdep/pull/436 which will now generate that help output based on the actual options given by `-h`. They looked like a copy and paste anyways.

wjwwood, 2016-02-14:
@dustingooding have a look at http://docs.ros.org/independent/api/rosdep/html/commands.html now (it should already be updated).

closed 2016-02-14 by wjwwood

dustingooding, 2016-02-14:
So is -R's previously described functionality (install package's rosdeps
and recursively the rosdeps of package's dependencies) the default
functionality?  And `-n` turns that off?   So, I should expect `rosdep
install metapackage_name` to install all the rosdeps of the entire "stack"?

wjwwood, 2016-02-15:
Well if you mean a catkin package, then no. You'd have to do `rosdep install --from-paths /path/to/folder/containing/metapackage`. But either way, yes I'd expect that `rosdep` installs the recursive dependencies unless you give it `-n`.