ros-infrastructure/rosdep #438: OsNotDetected

varocaraballo, 2016-03-07:
rosdep version: 0.11.4

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 545, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 466, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 147, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python2.7/dist-packages/rosdep2/__init__.py", line 85, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.py", line 50, in register_platforms
    register_linaro(context)
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.py", line 63, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 112, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python2.7/dist-packages/rospkg/os_detect.py", line 584, in get_name
    self.detect_os()
  File "/usr/lib/python2.7/dist-packages/rospkg/os_detect.py", line 554, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s"%attempted)
OsNotDetected: Could not detect OS, tried ['windows', 'ubuntu', 'rhel', 'qnx', 'osx', 'opensuse', 'opensuse', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'debian', 'cygwin', 'centos', 'arch']

tfoote, 2016-03-07:
Can you show the output of `lsb_release -a` on that system?

varocaraballo, 2016-03-08:
The output of lsb_release -a is:
No LSB modules are available.
Distributor ID:	elementary OS
Description:	elementary OS Freya
Release:	0.3.2
Codename:	freya

zhijieliu, 2016-05-06:
i meet the  same  problem 。ask  for help

tfoote, 2016-05-06:
This looks like there's a bug in the underlying OS detection https://github.com/ros-infrastructure/rospkg/blob/master/src/rospkg/os_detect.py

After that there probably needs to be an extended version aliasing extended in the elementary section here: https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/platforms/debian.py

This is not a platform that I know of core developers using so we can't easily test or maintain. Pull requests would be helpful.

zhijieliu, 2016-05-06:
I open the link but i do not kown what  is it and how can i solve this problem.  I am a new learner 、Can you explain it？Thanks very much.

tfoote, 2016-05-06:
To fix this you'll need to pull the package from source, patch it to work correctly and verify it. And hopefully submit a pull request so others can benefit from your work. 

If you're a just learning I highly recommend that you use Ubuntu, even if just in a VM, to get started. This is just going to be the first of many problem you run into using an unsupported OS. It's certainly possible but will require debugging a bunch of little things like this.

zhijieliu, 2016-05-06:
thanks very much. Your suggestion is very helpful、 I will  try to pull the package from source. thanks again！

tfoote, 2016-05-06:
It looks like this has been fixed in master, but not yet released: https://github.com/ros-infrastructure/rosdep/pull/408

deefunkt, 2016-05-21:
For anyone else struggling, an easy but kinda dodgy solution is to change the $ROS_OS_OVERRIDE variable to ubuntu and change the sources in /etc/apt/sources to work with ubuntu Trusty.

wjwwood, 2017-07-28:
I'm going to close this since it seems a workaround was found.

closed 2017-07-28 by wjwwood