ros-infrastructure/rosdep #440: --force-yes option

ZdenekM, 2016-03-23:
When e.g. mongodb is going to be installed, then only "-y" is not enough and "--force-yes" is required to suppress further questions - encountered on Travis.

tfoote, 2016-03-23:
You should not need the --force-yes. That suggests that you have not installed the signing key or something else is incorrectly configured. 

In general tools should not be using force options. Please open a question on answers.ros.org with more details.

closed 2016-03-23 by tfoote

jvgomez, 2016-05-19:
Actually, this is something I was looking for as well. I have a private apt repository and I try to avoid to sign the packages (just for simplicity). Without `--force-yes` I cannot install those packages with rosdep.

tfoote, 2016-05-19:
You don't need to sign the packages, just the repository. When you add the repository to the sources.list you just need to import the repository signing key. And setting up the repository with the signing key is a one time cost. Probably less than 15 minutes even if you have to create the gpg key. See here: https://wiki.debian.org/SettingUpSignedAptRepositoryWithReprepro

We certainly could consider it but I would strongly advocate not going this route. Installing packages without at least signing the repository means that it's completely vulnerable to a trivial man in the middle attack and the results will be installed with sudo permissions. It would be a large security risk.

jvgomez, 2016-05-23:
@tfoote Thanks for the info. I was not able install packages unless I signed them as well. In any case, it is just a command so the solution is still satisfactory. I was just avoiding that for simplicity but now I see that it is not much more complex :)

Again, thanks a lot!