ros-infrastructure/rosdep #443: "rosdep generate_bash" ?

k-okada, 2016-04-01:
Hi, we used have `rosdep generate_bash` (http://docs.ros.org/diamondback/api/rosdep/html/rosdep.main'-pysrc.html) which output a list of  all dependent deb packages, but seems removed now. What is the equivalent commands for that?
```
rosdep keys euslisp | xargs rosdep resolve | grep -v ^#
```
?

tfoote, 2016-04-01:
Does `rosdep install -s` not work for you?  You can use `--reinstall` to get all packages not just the ones needed for your system.

k-okada, 2016-04-02:
Thank you!!! `rosdep -s --reinstall` worked for me!
% sometimes, I have to explain to non-ros users, in this case, I'd like to install all dependent packages of X ...
```
$ rosdep install -s --reinstall assimp_devel
#[apt] Installation commands:
  sudo -H apt-get install ros-indigo-rosboost-cfg
  sudo -H apt-get install git-core
  sudo -H apt-get install ros-indigo-catkin
  sudo -H apt-get install zlib1g-dev
  sudo -H apt-get install openssl
  sudo -H apt-get install ros-indigo-mk
  sudo -H apt-get install ros-indigo-rosbuild
  sudo -H apt-get install ca-certificates
  sudo -H apt-get install unzip
  sudo -H apt-get install libboost-all-dev
```

closed 2016-04-02 by k-okada