ros-infrastructure/rosdep #445: rosdep check doesn't consider cmake-only packages

rhaschke, 2016-04-11:
Having a cmake-only package in my workspace
```xml
<export>
    <build_type>cmake</build_type>
</export>
```
rosdep complains that this package is not available:
`ERROR[urdf]: Cannot locate rosdep definition for [urdfdom-headers]`

Of course, I can use `--skip-keys "urdfdom-headers urdfdom"` to make rosdep ignore those packages.
However, I think rosdep should find them in the workspace.

wjwwood, 2016-04-12:
@rhaschke I guess you're using the "released" version? The upstream version of `urdfdom_headers` (at least) doesn't have a `package.xml`: https://github.com/ros/urdfdom_headers

rhaschke, 2016-04-12:
I'm compiling my own urdfdom + urdfdom_headers using a package.xml similar to the ones announced on the github project sites. catkin build successfully uses this package.xml to define the build order.
However, rosdep ignores the package.xml - probably, because it's a non-catkin package.

tfoote, 2016-04-12:
Can you provide more information how to reproduce your problem. rosdep does not care about the build_type of a package.

In particular I'm surprised by the name `urdfdom-headers`. Who is referencing it with a `-` in the middle? The upstream packaging uses an `_`.

wjwwood, 2016-04-12:
@rhaschke is it possible the `package.xml` name is `urdfdom_headers`, but `rosdep` wants `urdfdom-headers`? If so, can you change the name in the `package.xml` as a workaround?

I also don't see anywhere that `rosdep` cares about the `build_type`, though I may be missing it.

rhaschke, 2016-04-13:
@tfoote Well spotted. Indeed I was requesting a `build_depend` named `urdfdom-headers` although the package was named `urdfdom_headers`. However, the error remains, now with an `_`:
`ERROR[urdf]: Cannot locate rosdep definition for [urdfdom_headers]`

This isn't an urgent issue. Don't put too much effort into it.

mrjogo, 2016-12-21:
I'm running into the same issue when doing a full kinetic build from source. `rosdep check PACKAGE` fails with `ERROR: Rosdep cannot find all required resources to answer your query
Missing resource PACKAGE` for the following CMake packages: octomap, opencv3, orocos_kdl, python_orocos_kdl, and stage.

Using `--from-paths` pointed at the package share folder works.

mrjogo, 2016-12-21:
The core issue appears to be how `ROS_PACKAGE_PATH` is constructed by the environment hook script generated from `ros/env-hooks/10.ros.sh.em`.

The script takes the paths specified by the `CMAKE_PREFIX_PATH` environment variable, but then creates `ROS_PACKAGE_PATH` only from paths that contain a .catkin file. Workspaces that only contain packages with `<build_type>cmake</build_type>` will not be added to `ROS_PACKAGE_PATH`, which is used to find packages in `rospkg.RosPack` and in turn in `rosdep`. Note that as soon as a Catkin package is built in the same workspace, the problem is masked because a `.catkin` is created (I did all my testing using `catkin_make_isolated --install`)

It seems like `10.ros.sh` should check directly for a `package.xml` if `.catkin` is not present, but I'm not familiar enough with the system to know the ramifications of that.

A workaround if using `rospkg.RosPack` directly is to pass the constructor correct `ros_paths`.

Ryanf55, 2025-01-21:
This is still an issue on ROS 2 humble. I have a package that has the build type set to `cmake`. It gets installed by colcon into an underlay in `/opt/my_underlay/install/my_package`, but without a package.xml because it doesn't have one or need one.

Then, I `source `/opt/my_underlay/install/setup.bash` and run `rosdep install --from-paths src --ignore-src` with a package that expresses a dependency on `my_package`, rosdep will fail.

tfoote, 2025-01-21:
> This is a huge limitation of rosdep because it means we need to add skip keys for every single CMake package that we compile from source in our underlay. I just found this issue, but would love some suggestions on this because it's been a maintenance burden for us to skip keys or build them outside the colcon workspace manually with CMake.

The tool needs a compact way to know what's on your system and represents the dependency that is needed in your workspace.

> After sourcing the ws1 underlay, you can see that CMAKE_PREFIX_PATH has the path to my_pkg1's install directly, so the build will pass, but rosdep won't resolve the key.

This happens because you install configuration files that tell CMake where to search. aka `*.cmake` file or an env file to extend the `CMAKE_PREFIX_PATH` 

rosdep needs a similar indicator to be able to know about the presence of a package. There are two categories of packages ones in the rosdep database that are known and managed by the package manager on your platform. And there are those in your workspace that are built from source and have a marker file that lets rosdep know that they're there with the canonical name, ala the package.xml file. This is how rosdep knows that when a package depends on `urdfdom_headers` something is providing it on your system. By placing the package.xml file there the name declared, rosdep knows the package is there. Without that it can't know that the contents of `urdfdom_headers` is available. 

The simplest approach is to just add the package.xml file upstream which helps define the package reference. The next step is to make a simple intermediate package that depends on the upstream one confirms it's installed, and then adds the package.xml file so rosdep knows it's there. Or you can make a vendor package which integrates the installation and adds the package.xml files appropriately, and if it finds the dependency already on the system it can just set the appropriate package.xml to let rosdep know that the content is already there. 

The above reference urdfdom package has now started shipping the package.xml files as of Jazzy at least: https://packages.ubuntu.com/jammy/amd64/liburdfdom-dev/filelist  as one of the few core dependencies with this issue. 

Someone needs to provide rosdep with the appropriate metadata to be able to answer the query. A pure cmake package can do this with one line like this: https://github.com/ros2/orocos_kinematics_dynamics/blob/f48062a37e6697a3c1a96e3aecb4ab6c4c28a872/orocos_kdl/CMakeLists.txt#L137  with integrated source

Or a vendor package like this: https://github.com/ros2/tinydir_vendor/blob/f1ee71b2fb02b24ee40123a905b16922938ca80a/CMakeLists.txt#L78-L80