ros-infrastructure/rosdep #448: Rosdep experienced an error: 'module' object has no attribute 'sslwrap'

nagaban2, 2016-04-26:
The output for my "rosdep update" command resulted in:
ERROR: Rosdep experienced an error: 'module' object has no attribute 'sslwrap'
Please go to the rosdep page [1] and file a bug report with the stack trace below.

------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 545, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 446, in update_sources_list
    rosdep_data = download_rosdep_data(source.url)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 294, in download_rosdep_data
    f = urlopen(url, timeout=DOWNLOAD_TIMEOUT)
  File "/usr/lib/python2.7/urllib2.py", line 127, in urlopen
    return _opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 404, in open
    response = self._open(req, data)
  File "/usr/lib/python2.7/urllib2.py", line 422, in _open
    '_open', req)
  File "/usr/lib/python2.7/urllib2.py", line 382, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 1222, in https_open
    return self.do_open(httplib.HTTPSConnection, req)
  File "/usr/lib/python2.7/urllib2.py", line 1181, in do_open
    h.request(req.get_method(), req.get_selector(), req.data, headers)
  File "/usr/lib/python2.7/httplib.py", line 979, in request
    self._send_request(method, url, body, headers)
  File "/usr/lib/python2.7/httplib.py", line 1013, in _send_request
    self.endheaders(body)
  File "/usr/lib/python2.7/httplib.py", line 975, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 835, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 797, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 1182, in connect
    self.sock = ssl.wrap_socket(sock, self.key_file, self.cert_file)
  File "/usr/lib/python2.7/ssl.py", line 487, in wrap_socket
    ciphers=ciphers)
  File "/usr/lib/python2.7/ssl.py", line 238, in __init__
    self._sslobj = _ssl.sslwrap(self._sock, server_side,
AttributeError: 'module' object has no attribute 'sslwrap'

wjwwood, 2016-04-26:
What OS are you on and what version of Python 2.7 are you using. According to some posts on the internet SSL might be broken on 2.7.9 for example:

- https://github.com/eventlet/eventlet/issues/135
- http://bugs.python.org/issue22438

nagaban2, 2016-04-26:
Ubuntu 14.04, Python 2.7.10 , Anaconda 2.3.0

The link says SSL is broken for Python >= 2.7.9, so that makes sense. Haven't found the solution yet, but I'll look around. Let me know if you already know how to solve it! Thanks

wjwwood, 2017-07-28:
I think the solution is to use a newer version of Python (with fixed SSL). I'm going to close this for now, since there's nothing we can do in rosdep to resolve this. This question might be better to put on answers.ros.org or just stackoverflow if it is still an issue.

closed 2017-07-28 by wjwwood