ros-infrastructure/rosdep #452: rosdep cannot init on xenial without ca-certificates installed

tfoote, 2016-05-06:
It init's from the https url, and if it does not have ca-certificates it interprets the errors as the website is down:

```
tfoote@yeti:~/work/osrf/docker_images/ros/jade/jade-ros-core Last: [0] (87s Seconds)
$ docker run -ti kinetic-core 
root@93885206a51a:/# rosdep init
ERROR: cannot download default sources list from:
https://raw.github.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.
```

Adding the ca-certificates lets it continue:
```
root@93885206a51a:/# apt-get update && apt-get install ca-certificates -q
Get:1 http://packages.ros.org/ros/ubuntu xenial InRelease [4,008 B]
Get:2 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages [100 kB]
Get:3 http://archive.ubuntu.com/ubuntu xenial InRelease [247 kB]
Get:4 http://archive.ubuntu.com/ubuntu xenial-updates InRelease [93.3 kB]
Get:5 http://archive.ubuntu.com/ubuntu xenial-security InRelease [93.3 kB]
Get:6 http://archive.ubuntu.com/ubuntu xenial/main Sources [1,103 kB]
Get:7 http://archive.ubuntu.com/ubuntu xenial/restricted Sources [5,179 B]
Get:8 http://archive.ubuntu.com/ubuntu xenial/universe Sources [9,802 kB]
Get:9 http://archive.ubuntu.com/ubuntu xenial/main amd64 Packages [1,558 kB]
Get:10 http://archive.ubuntu.com/ubuntu xenial/restricted amd64 Packages [14.1 kB]
Get:11 http://archive.ubuntu.com/ubuntu xenial/universe amd64 Packages [9,827 kB]
Get:12 http://archive.ubuntu.com/ubuntu xenial-updates/main Sources [16.9 kB]
Get:13 http://archive.ubuntu.com/ubuntu xenial-updates/restricted Sources [40 B]
Get:14 http://archive.ubuntu.com/ubuntu xenial-updates/universe Sources [1,989 B]
Get:15 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages [55.5 kB]
Get:16 http://archive.ubuntu.com/ubuntu xenial-updates/restricted amd64 Packages [40 B]
Get:17 http://archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages [12.0 kB]
Get:18 http://archive.ubuntu.com/ubuntu xenial-security/main Sources [9,434 B]
Get:19 http://archive.ubuntu.com/ubuntu xenial-security/restricted Sources [40 B]
Get:20 http://archive.ubuntu.com/ubuntu xenial-security/universe Sources [1,491 B]
Get:21 http://archive.ubuntu.com/ubuntu xenial-security/main amd64 Packages [42.1 kB]
Get:22 http://archive.ubuntu.com/ubuntu xenial-security/restricted amd64 Packages [40 B]
Get:23 http://archive.ubuntu.com/ubuntu xenial-security/universe amd64 Packages [6,767 B]
Fetched 23.0 MB in 3s (6,433 kB/s)    
Reading package lists... Done
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  openssl
The following NEW packages will be installed:
  ca-certificates openssl
0 upgraded, 2 newly installed, 0 to remove and 51 not upgraded.
Need to get 682 kB of archives.
After this operation, 1,453 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 openssl amd64 1.0.2g-1ubuntu4.1 [491 kB]
Get:2 http://archive.ubuntu.com/ubuntu xenial/main amd64 ca-certificates all 20160104ubuntu1 [191 kB]
Fetched 682 kB in 0s (42.5 MB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package openssl.
(Reading database ... 9783 files and directories currently installed.)
Preparing to unpack .../openssl_1.0.2g-1ubuntu4.1_amd64.deb ...
Unpacking openssl (1.0.2g-1ubuntu4.1) ...
Selecting previously unselected package ca-certificates.
Preparing to unpack .../ca-certificates_20160104ubuntu1_all.deb ...
Unpacking ca-certificates (20160104ubuntu1) ...
Setting up openssl (1.0.2g-1ubuntu4.1) ...
Setting up ca-certificates (20160104ubuntu1) ...
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76.)
debconf: falling back to frontend: Readline
debconf: unable to initialize frontend: Readline
debconf: (Can't locate Term/ReadLine.pm in @INC (you may need to install the Term::ReadLine module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.22.1 /usr/local/share/perl/5.22.1 /usr/lib/x86_64-linux-gnu/perl5/5.22 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.22 /usr/share/perl/5.22 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base .) at /usr/share/perl5/Debconf/FrontEnd/Readline.pm line 7.)
debconf: falling back to frontend: Teletype
Processing triggers for ca-certificates (20160104ubuntu1) ...
Updating certificates in /etc/ssl/certs...
173 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
root@93885206a51a:/# rosdep init
Wrote /etc/ros/rosdep/sources.list.d/20-default.list
Recommended: please run

	rosdep update

root@93885206a51a:/# 
```

I discovered this when testing adapting the ros docker builds on xenial: https://github.com/osrf/docker_images/blob/master/ros/jade/jade-ros-core/Dockerfile

As of xenial ca-certificates is now required to access the https urls since in Python 2.7.9 they have made the default ssl usage actually secure: https://www.python.org/downloads/release/python-279/

furushchev, 2016-10-13:
Does this update solve certificate problem?
I tried to `rosdep update` on my 16.04 / kinetic, and still occurs the same error:

```bash
$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml]:
        <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml]:
        <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml]:
        <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml]:
        <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)> (https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml)
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml]:
        Failed to download target platform data for gbpdistro:
        <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)>
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
ERROR: error loading sources list:
        <urlopen error <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)> (https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml)>
```

```bash
$ sudo apt show python-rosdep
Package: python-rosdep
Version: 0.11.5-1
Priority: optional
Section: python
Maintainer: Tully Foote, Ken Conley <tfoote@osrfoundation.org>
Installed-Size: 289 kB
Depends: python (>= 2.7), python (<< 2.8), python:any (>= 2.7.1-0ubuntu2), python-catkin-pkg, python-rospkg (>= 1.0.37), python-rosdistro (>= 0.4.0), python-yaml, ca-certificates
Conflicts: python3-rosdep
Download-Size: 44.6 kB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros-shadow-fixed/ubuntu xenial/main arm64 Packages
Description: rosdep package manager abstrction tool for ROS
 Command-line tool for installing system dependencies on a variety of platforms.

N: There is 1 additional record. Please use the '-a' switch to see it
```

tfoote, 2016-10-13:
@furushchev I see arm64 in your traceback are you on a TX1? If so see [this issue](https://github.com/ros/rosdistro/issues/12894#issuecomment-251773179)

furushchev, 2016-10-14:
@tfoote I saw the issue you mentioned, and now it works! This is a TX1 Problem... Anyway thank you very much!

tfoote, 2016-10-14:
fixed in #454

closed 2016-10-14 by tfoote