ros-infrastructure/rosdep #455: http://www.ros.org/wiki/rosdep

Abou-Hussein, 2016-05-19:
Traceback (most recent call last):
  File "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/main.py", line 353, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/main.py", line 447, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/main.py", line 654, in command_install
    installer.install(uninstalled, **install_options)
  File "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/installers.py", line 485, in install
    verbose=verbose, quiet=quiet)
  File "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/installers.py", line 515, in install_resolved
    command = installer.get_install_command(resolved, interactive=interactive, reinstall=reinstall, quiet=quiet)
  File "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/platforms/osx.py", line 316, in get_install_command
    resolved = self.remove_duplicate_dependencies(resolved)
  File "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/platforms/osx.py", line 347, in remove_duplicate_dependencies
    if brew_strip_pkg_name(other.package) == brew_strip_pkg_name(d):
  File "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/platforms/osx.py", line 157, in brew_strip_pkg_name
    return package.split('/')[-1]
TypeError: a bytes-like object is required, not 'str'

tfoote, 2016-05-19:
Please provide enough information to reproduce your error. 

It looks like you're running in an anaconda environment. We've seen some minor issues with tools working inside anaconda environments. It's not a supported platform so we rely on community contributions to help broaden our support. We run continuous integration on python 2.6, 2.7, and 3.4 at the moment.

Abou-Hussein, 2016-05-19:
I was trying to install ROS indigo on my mac OS-X El Capitan. I was
following the following instructions in the page below:

   - http://wiki.ros.org/indigo/Installation/OSX/Homebrew/Source#Setup

I met some errors but resolved them as they arouse. Till I met this command
which is responsible for resolving dependencies for creating the catkin
workspace: rosdep install --from-paths src --ignore-src --rosdistro indigo
-y

Then I met the error message I sent, here it is again below:

   -

   @Abou-Hussein:~/ros_catkin_ws$ rosdep install --from-paths src
   --ignore-src --rosdistro indigo -y

   ERROR: Rosdep experienced an error: a bytes-like object is required, not
   'str'

   Please go to the rosdep page [1] and file a bug report with the stack
   trace below.

   [1] : http://www.ros.org/wiki/rosdep

   rosdep version: 0.11.4

   Traceback (most recent call last):

     File
   "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/main.py",
   line 136, in rosdep_main

       exit_code = _rosdep_main(args)

     File
   "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/main.py",
   line 353, in _rosdep_main

       return _package_args_handler(command, parser, options, args)

     File
   "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/main.py",
   line 447, in _package_args_handler

       return command_handlers[command](lookup, packages, options)

     File
   "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/main.py",
   line 654, in command_install

       installer.install(uninstalled, **install_options)

     File
   "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/installers.py",
   line 485, in install

       verbose=verbose, quiet=quiet)

     File
   "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/installers.py",
   line 515, in install_resolved

       command = installer.get_install_command(resolved,
   interactive=interactive, reinstall=reinstall, quiet=quiet)

     File
   "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/platforms/osx.py",
   line 316, in get_install_command

       resolved = self.remove_duplicate_dependencies(resolved)

     File
   "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/platforms/osx.py",
   line 347, in remove_duplicate_dependencies

       if brew_strip_pkg_name(other.package) == brew_strip_pkg_name(d):

     File
   "/Users/Abou-Hussein/anaconda/lib/python3.5/site-packages/rosdep2/platforms/osx.py",
   line 157, in brew_strip_pkg_name

       return package.split('/')[-1]

   TypeError: a bytes-like object is required, not 'str'

I hope you can help me, this is critical for work.
Thank you in advance. Awaiting your reply.

wjwwood, 2016-05-19:
@Abou-Hussein you're using Python3, which is not supported for ROS 1 on OS X (that's not to say it couldn't work we just don't spend time working on it).

I'd recommend making sure you're using Python2 everywhere and especially making sure your `pip` points to Python2 and not Python3.

jon-courtney, 2017-06-06:
In running the `rosdep` command above, it seems that the `boost` package is somehow picked up as an array of bytes rather than as a string.  I was able to get past this point by changing the body of `brew_strip_pkg_name(package)` in `osx.py` to the following:
```
    if type(package) is bytes:
        package = package.decode('utf-8')

    return package.split('/')[-1]
```

wjwwood, 2017-07-28:
I believe this was fixed by https://github.com/ros-infrastructure/rosdep/pull/441.

closed 2017-07-28 by wjwwood