ros-infrastructure/rosdep #456: pip doesn't find python packages installed in /opt/ros/indigo

asmodehn, 2016-05-23:
I haven't found the root cause of this problem yet, however this is my current case, which I find quite confusing (coming from a python background) : 

```
alexv@asmodehn:~$ python -c "import tornado; print tornado.__file__; print tornado.version"
/usr/lib/python2.7/dist-packages/tornado/__init__.pyc
3.1.1
alexv@asmodehn:~$ pip freeze | grep tornado
tornado==3.1.1
alexv@asmodehn:~$ source /opt/ros/indigo/setup.bash 
alexv@asmodehn:~$ python -c "import tornado; print tornado.__file__; print tornado.version"
/opt/ros/indigo/lib/python2.7/dist-packages/tornado/__init__.pyc
4.0.2
alexv@asmodehn:~$ pip freeze | grep tornado
tornado==3.1.1
```

asmodehn, 2016-05-23:
rosdep installs python ros packages in /opt/ros/indigo (through ros debian packages), so maybe this is more of a "package structure" issue rather than rosdep itself ?

asmodehn, 2016-05-25:
After more research, pip actually finds packages in /opt/ros/indigo : 

```
alexv@asmodehn:~$ pip list | grep ros
pyros-setup (0.1.0)
ros-buildfarm (1.1.0)
rosdep (0.11.4)
rosdistro (0.4.7)
rospkg (1.0.39)
```

```
alexv@asmodehn:~$ source /opt/ros/indigo/setup.bash 
alexv@asmodehn:~$ pip list | grep ros
gazebo-ros (2.4.10)
pyros-setup (0.0.8)
pyros-test (0.0.3)
ros-buildfarm (1.1.0)
rosapi (0.7.14)
rosbag (1.11.19)
rosboost-cfg (1.11.12)
rosbridge-library (0.7.14)
rosbridge-server (0.7.14)
rosclean (1.11.12)
roscreate (1.11.12)
rosdep (0.11.4)
rosdistro (0.4.7)
rosdoc-lite (0.2.6)
rosgraph (1.11.19)
roslaunch (1.11.19)
roslib (1.11.12)
roslint (0.10.0)
roslz4 (1.11.19)
rosmake (1.11.12)
rosmaster (1.11.19)
rosmsg (1.11.19)
rosnode (1.11.19)
rosparam (1.11.19)
rospkg (1.0.39)
rospy (1.11.19)
rospy-message-converter (0.4.0)
rosservice (1.11.19)
rostest (1.11.19)
rostopic (1.11.19)
rosunit (1.11.12)
roswtf (1.11.19)
smach-ros (2.0.0)
tf2-ros (0.5.13)
alexv@asmodehn:~$ pip freeze | grep tornado
tornado==3.1.1
alexv@asmodehn:~$ pip list | grep tornado
tornado (3.1.1)
```

So maybe the problem comes from somewhere else ?

tfoote, 2016-05-25:
I'd suggest digging into how pip interacts with the PYTHONPATH. It may privilege things on standard paths over things inserted on the PYTHONPATH

asmodehn, 2017-04-03:
@tfoote From my understanding PYTHONPATH is always a "last step" add on at the beginning of the sys.path list of paths where python modules/packages can be found. BTW this is different from the way ROS treat PYTHONPATH as the stack of workspaces (which kind of was decided against in python environment - venv alone, or system and venv, but no venv stacking mechanism).
Also I wouldn't expect pip to behave in a different way than python does.

I digged a bit more...
It seems this doesn't happen with a recent version of pip and a recent version of tornado package : 

```
alexv@AlexV-Linux:~$ pip --version
pip 8.1.2 from /usr/local/lib/python2.7/dist-packages (python 2.7)
alexv@AlexV-Linux:~$ pip list | grep ros
pyros-config (0.2.0)
pyros-setup (0.2.0)
rosdep (0.11.5)
rosdistro (0.6.1)
rosdistro-modules (0.6.1)
rosinstall (0.7.8)
rosinstall-generator (0.1.13)
rospkg (1.1.0)
rospkg-modules (1.1.0)
You are using pip version 8.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

alexv@AlexV-Linux:~$ source /opt/ros/indigo/setup.bash 
alexv@AlexV-Linux:~$ pip list | grep ros
gazebo-ros (2.4.13)
pyros-config (0.2.0)
pyros-setup (0.2.0)
pyros-test (0.0.6)
pyros-utils (0.1.3)
rosbag (1.11.21)
rosboost-cfg (1.11.14)
rosclean (1.11.14)
roscreate (1.11.14)
rosdep (0.11.5)
rosdistro (0.6.1)
rosdistro-modules (0.6.1)
rosgraph (1.11.21)
rosinstall (0.7.8)
rosinstall-generator (0.1.13)
roslaunch (1.11.21)
roslib (1.11.14)
roslint (0.10.0)
roslz4 (1.11.21)
rosmake (1.11.14)
rosmaster (1.11.21)
rosmsg (1.11.21)
rosnode (1.11.21)
rosparam (1.11.21)
rospkg (1.1.0)
rospkg-modules (1.1.0)
rospy (1.11.21)
rospy-message-converter (0.4.0)
rosservice (1.11.21)
rostest (1.11.21)
rostopic (1.11.21)
rosunit (1.11.14)
roswtf (1.11.21)
tf2-ros (0.5.15)
You are using pip version 8.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```
```
alexv@AlexV-Linux:~$ pip list | grep tornado
tornado (4.2.1)
You are using pip version 8.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
alexv@AlexV-Linux:~$ python -c "import tornado; print tornado.__file__"
/opt/ros/indigo/lib/python2.7/dist-packages/tornado/__init__.pyc
alexv@AlexV-Linux:~$ pip freeze | grep tornado
You are using pip version 8.1.2, however version 9.0.1 is available.
[... 184 lines cut ...]
Selecting previously unselected package ros-indigo-rosauth.
Preparing to unpack .../ros-indigo-rosauth_0.1.7-0trusty-20170313-093021-0700_amd64.deb ...
Unpacking ros-indigo-rosauth (0.1.7-0trusty-20170313-093021-0700) ...
Selecting previously unselected package ros-indigo-rosbridge-server.
Preparing to unpack .../ros-indigo-rosbridge-server_0.7.16-0trusty-20170313-104800-0700_amd64.deb ...
Unpacking ros-indigo-rosbridge-server (0.7.16-0trusty-20170313-104800-0700) ...
Setting up python-bson (2.6.3-1build1) ...
Setting up python-bson-ext (2.6.3-1build1) ...
Setting up ros-indigo-rosbridge-library (0.7.16-0trusty-20170313-095924-0700) ...
Setting up ros-indigo-rosapi (0.7.16-0trusty-20170313-104233-0700) ...
Setting up ros-indigo-rosauth (0.1.7-0trusty-20170313-093021-0700) ...
Setting up ros-indigo-rosbridge-server (0.7.16-0trusty-20170313-104800-0700) ...
alexv@AlexV-Linux:~$ python -c "import tornado; print tornado.__file__; print tornado.version"
/opt/ros/indigo/lib/python2.7/dist-packages/tornado/__init__.pyc
4.0.2
alexv@AlexV-Linux:~$ pip freeze | grep tornado
tornado==3.1.1
```
And the problem is here again.

So apparently there is some ROS packaging limitation when used with multiple python packages inside one ROS package ?
Something that should not be abused and we should prevent against ?
How can we confirm that ??

In anycase it doesn't seem related to rosdep, so we should find the root cause and recreate an issue in the appropriate repo.

wjwwood, 2017-07-28:
Since this doesn't appear to be related to how `rosdep` uses `pip`, I'll close this for now. It can be reopened if need be.

closed 2017-07-28 by wjwwood