ros-infrastructure/rosdep #457: Rosdep Error: UnicodeDecodeError

allenh1, 2016-05-24:
Ehm... Gentoo is not having fun with python 3.4, it would seem. No idea what happened here... Rosdep appears to be convinced this is a bug, so I will definitely file a bug report, but... I'm not so sure. Likely a python 3.4 incompatibility thing.

```
allenh1@localhost ~/ros_catkin_ws $ rosdep install --from-paths src --ignore-src --rosdistro kinetic -y

ERROR: Rosdep experienced an error: 'ascii' codec can't decode byte 0xc5 in position 366: ordinal not in range(128)
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.4

Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib64/python3.4/site-packages/rosdep2/main.py", line 353, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib64/python3.4/site-packages/rosdep2/main.py", line 406, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib64/python3.4/site-packages/rosdep2/catkin_packages.py", line 33, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib64/python3.4/site-packages/catkin_pkg/packages.py", line 83, in find_packages
    packages = find_packages_allowing_duplicates(basepath, exclude_paths=exclude_paths, exclude_subspaces=exclude_subspaces, warnings=warnings)
  File "/usr/lib64/python3.4/site-packages/catkin_pkg/packages.py", line 110, in find_packages_allowing_duplicates
    packages[path] = parse_package(os.path.join(basepath, path), warnings=warnings)
  File "/usr/lib64/python3.4/site-packages/catkin_pkg/package.py", line 370, in parse_package
    return parse_package_string(f.read(), filename, warnings=warnings)
  File "/usr/lib64/python3.4/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc5 in position 366: ordinal not in range(128)
```

allenh1, 2016-05-24:
Well, maybe not? I changed the system to use 2.7 instead. Now it doesn't like pickles. 

```
allenh1@localhost ~/ros_catkin_ws $ rosdep install --from-paths src --ignore-src --rosdistro kinetic -y

ERROR: Rosdep experienced an error: line:
	yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx
unsupported pickle protocol: 4
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.4

Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib64/python2.7/site-packages/rosdep2/main.py", line 353, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib64/python2.7/site-packages/rosdep2/main.py", line 434, in _package_args_handler
    lookup = _get_default_RosdepLookup(options)
  File "/usr/lib64/python2.7/site-packages/rosdep2/main.py", line 127, in _get_default_RosdepLookup
    verbose=options.verbose)
  File "/usr/lib64/python2.7/site-packages/rosdep2/sources_list.py", line 607, in create_default
    sources = load_cached_sources_list(sources_cache_dir=sources_cache_dir, verbose=verbose)
  File "/usr/lib64/python2.7/site-packages/rosdep2/sources_list.py", line 508, in load_cached_sources_list
    return parse_sources_data(cache_data, origin=cache_index, model=model)
  File "/usr/lib64/python2.7/site-packages/rosdep2/sources_list.py", line 366, in parse_sources_data
    raise InvalidData("line:\n\t%s\n%s"%(line, e), origin=origin)
InvalidData: line:
	yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx
unsupported pickle protocol: 4
```

wjwwood, 2016-05-24:
If you switch between python2 and python3 you have to delete the cache files.

allenh1, 2016-05-24:
Hm. Ok. 

```
allenh1@localhost ~/ros_catkin_ws $ rm -rf ~/.ros/rosdep/sources.cache
allenh1@localhost ~/ros_catkin_ws $ rosdep update
```

this did the trick, as you suggested. So I take it this is a python 3 incompatibility?

wjwwood, 2016-05-24:
I don't know what the original issue is (I don't have time to look into it at the moment), but I did recognized the pickle error. If you're interested in that, you can take a look in the existing issues. It has been discussed before, but I don't remember where exactly.

allenh1, 2016-05-24:
Will do.

NikolausDemmel, 2016-05-24:
To avoid the pickle error, we could specify pickle protocol to always be `2` instead of ["latest" as is currently](https://github.com/ros-infrastructure/rosdep/blob/d9adc2f6169966ae8c910830284397159a57e4a9/src/rosdep2/sources_list.py#L535). Protocol `2` is the latest on python 2.7. Even with that additional work might be needed, since e.g. strings from python 3 will decode as unicode in Python 2, and rosdep might not like that either.

I haven't looked into the original issue with python 3 compatibility yet.

allenh1, 2016-05-24:
Would it be possible to just catch this as an exception and, if it is thrown, remove the old cash and attempt to perform a rosdep update again?

NikolausDemmel, 2016-05-24:
That would be a good first step. Though it might maybe be more sensible to simply output a more informative error message with instructions on how to fix the issue instead of running `update` automagically. I would find it a bit odd if `rosdep check ...` might suddenly call update under the hood in this error case.

tfoote, 2016-05-25:
Another option would be to have python2 and python3 drop the caches in non-colliding locations. It would be a little less coherent if you tried to update with one version and install with a different version. But it's better to miss the cache than traceback, and it could detect the other version and provide instructions to mention that you need to use a consistent python version.

allenh1, 2016-05-26:
@tfoote that makes sense to me. Though I think the error message should be the first thing to change. Maybe just an addition at the end of it reading something like "are you using the same version of Python you used to initialize?"

allenh1, 2016-05-26:
One more thought: what if we just store the Python version with the databases and check that the current python version is equivalent. If it isn't, rosdep should exit.

wjwwood, 2017-07-28:
@allenh1 should this be closed along with #459 or are you going to open a new pr?

allenh1, 2017-07-28:
I could fix this in the future, if requested. But I don't think this is particularly high priority.

allenh1, 2017-08-04:
@wjwwood I'm going to close this one.

closed 2017-08-04 by allenh1