ros-infrastructure/rosdep #460: error while installing ros using rosdep

robofollow, 2016-05-31:
i am trying to install ROS Indigo in Utopic so i tried to install it from source... but i get the following error when i execute the command 
rosdep install --from-paths src --ignore-src --rosdistro indigo -y

**ERROR: Rosdep experienced an error: rosdep data for [libpcl-all-dev] must be a dictionary
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Traceback (most recent call last):
File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
exit_code = rosdep_main(args)
File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 353, in rosdep_main
return _package_args_handler(command, parser, options, args)
File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 447, in _package_args_handler
return command_handlerscommand
File "/usr/local/lib/python2.7/dist-packages/rosdep2/main.py", line 631, in command_install
uninstalled, errors = installer.get_uninstalled(packages, implicit=options.recursive, verbose=options.verbose)
File "/usr/local/lib/python2.7/dist-packages/rosdep2/installers.py", line 409, in get_uninstalled
resolutions, errors = self.lookup.resolve_all(resources, installer_context, implicit=implicit)
File "/usr/local/lib/python2.7/dist-packages/rosdep2/lookup.py", line 383, in resolve_all
rosdep_keys = self.get_rosdeps(resource_name, implicit=implicit)
File "/usr/local/lib/python2.7/dist-packages/rosdep2/lookup.py", line 305, in get_rosdeps
return self.loader.get_rosdeps(resource_name, implicit=implicit)
File "/usr/local/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py", line 127, in get_rosdeps
m = self._rospack.get_manifest(resource_name)
File "/usr/local/lib/python2.7/dist-packages/rospkg/rospack.py", line 164, in get_manifest
return self._load_manifest(name)
File "/usr/local/lib/python2.7/dist-packages/rospkg/rospack.py", line 208, in _load_manifest
retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
File "/usr/local/lib/python2.7/dist-packages/rospkg/manifest.py", line 393, in parse_manifest_file
_static_rosdep_view = init_rospack_interface()
File "/usr/local/lib/python2.7/dist-packages/rosdep2/rospack.py", line 59, in init_rospack_interface
return lookup.get_rosdep_view(DEFAULT_VIEW_KEY)
File "/usr/local/lib/python2.7/dist-packages/rosdep2/lookup.py", line 605, in get_rosdep_view
view = self.create_rosdep_view(view_key, dependencies + [view_key], verbose=verbose)
File "/usr/local/lib/python2.7/dist-packages/rosdep2/lookup.py", line 550, in create_rosdep_view
view.merge(db_entry, verbose=verbose)
File "/usr/local/lib/python2.7/dist-packages/rosdep2/lookup.py", line 211, in merge
update_definition = RosdepDefinition(dep_name, dep_data, update_entry.origin)
File "/usr/local/lib/python2.7/dist-packages/rosdep2/lookup.py", line 65, in __init
raise InvalidData("rosdep data for [%s] must be a dictionary" % (self.rosdep_key), origin=origin)
InvalidData: rosdep data for [libpcl-all-dev] must be a dictionary**

I have followed the instructions given in this thread
 https://gist.github.com/nuclearmistake/86ec610b8ebf1bb3df60
but i get stuck in the rosdep install command.
btw i have already installed gazebo5
where am i going wrong??

wjwwood, 2016-05-31:
It looks like there is an error in your "local" rosdep file, which is introduced here: https://gist.github.com/nuclearmistake/86ec610b8ebf1bb3df60#file-indigo_utopic_instructions-L18

Also, @trainman419 commented (https://gist.github.com/nuclearmistake/86ec610b8ebf1bb3df60#gistcomment-1403729) that the line might be incorrect. I think you'd get more eyes on your problem by asking on answers.ros.org.

I don't believe this is a rosdep bug, so I'm going to close it. I'll reopen it if that is necessary.

closed 2016-05-31 by wjwwood

robofollow, 2016-05-31:
i have seen that comment already and corrected it to "yaml" but i still get the error.

wjwwood, 2016-05-31:
I'd recommend you open a question on answers.ros.org and post the contents of the local yaml file.