ros-infrastructure/rosdep #464: ValueError: No JSON object could be decoded

Numero7, 2016-07-04:
ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Bad installer [homebrew]: Error while parsing brew info for 'tango-icon-theme'
 * Output of `brew info tango-icon-theme --json=v1`:
 
 * Error while parsing:
 Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/rosdep2/platforms/osx.py", line 184, in is_installed
    pkg_info = json.loads(std_out)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

NikolausDemmel, 2016-07-04:
Looks like https://github.com/ros-infrastructure/rosdep/issues/339

Can you try `brew update` and `brew tap --repair` as suggested in that issue?

Numero7, 2016-07-06:
In the meantime I actually nuked down brew and started from a fresh brew installation and it worked perfectly,
thanks for the advice.

closed 2016-07-06 by NikolausDemmel