ros-infrastructure/rosdep #465: Unable to locate package shiboken2 and libpyside2-dev

Thomio-Watanabe, 2016-07-05:
I have a fresh Ubuntu Xenial 16.04 installation, I'm trying to install ROS Kineckit from source and it cannot find libpyside2-dev, shiboken2 nor libshiboken2-dev.
I get the following error when running **rosdep install --from-paths src --ignore-src --rosdistro kinetic -y**:

    E: Unable to locate package shiboken2
    ERROR: the following rosdeps failed to install
        apt: command [sudo -H apt-get install -y shiboken2] failed

    E: Unable to locate package libpyside2-dev
    ERROR: the following rosdeps failed to install
        apt: command [sudo -H apt-get install -y libpyside2-dev] failed

How can I install these packages ?

wjwwood, 2016-07-05:
@Thomio-Watanabe those packages, as far as I know, are custom built and in our deb repository (packages.ros.org), so if you don't have that added to your apt sources, then those packages will not be available. That's my best guess anyways. @dirk-thomas are my assumptions correct?

dirk-thomas, 2016-07-05:
Yes, see https://github.com/ros-infrastructure/reprepro-updater/blob/master/config/pyside2.prerelease.yaml

Thomio-Watanabe, 2016-07-05:
Sorry guys, my bad. 
Adding the packages.ros.org repository solves the problem.
Thanks

closed 2016-07-05 by Thomio-Watanabe

Matthew-Wisniewski, 2016-08-08:
I know this has already been closed, but I've also got a fresh Ubuntu Xenial 16.04 installation, and am trying to install ROS kinetic from source and I'm getting the same error: 

E: Unable to locate package libpyside2-dev
ERROR: the following rosdeps failed to install
    apt: command [sudo -H apt-get install -y libpyside2-dev] failed

To add the packages.ros.org repository I've entered the command below, but then when I try to install again I'm still getting the same error. Am I missing something?:

sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'

dirk-thomas, 2016-08-08:
The Debian packages for Shiboken2 / PySide2 have been imported from a PPA and are in the ROS package repos: please see [Xenial(http://packages.ros.org/ros/ubuntu/dists/xenial/main/binary-amd64/Packages) where you will find `libpyside2-dev`. Have you run `sudo apt- update`? Does it succesfully pull the information from `packages.ros.org`?

Matthew-Wisniewski, 2016-08-09:
I ran the sudo apt- update, tried again and it successfully installed the package. Such a rookie mistake... :P I'm running on an RPi3, am new to linux and was a bit hesitant to update at first because I thought I could mess up the Xenial installation by updating to the bleeding edge. Thanks for the help.

lvibb, 2017-06-07:
try:
sudo -H apt-get install shiboken2 **--allow-unauthenticated**

tfoote, 2017-06-07:
@lvibb That's highly not recommended. If that fixed your system you should check your sources and make sure to authorize the correct gpg keys.

gaoethan, 2017-06-14:
I resolved my issue with the following operation:
**$** sudo add-apt-repository ppa:thopiekar/pyside-git
**$** sudo apt update
**$** rosdep install --from-paths src --ignore-src --rosdistro kinetic -y

dirk-thomas, 2017-06-14:
@gaoethan Adding the PPA should not be necessary if you are using an up-to-date packages.ros.org. What platform are you on?

gaoethan, 2017-06-16:
@dirk-thomas My OS is 64-bit Ubuntu 16.04.2 LTS

tfoote, 2017-06-16:
@gaoethan What does `apt-cache policy libpyside2-dev` show? As well as `cat /etc/apt/sources.list.d/ros-latest.list` ?

gaoethan, 2017-06-16:
@tfoote Initially there is only libpyside1.2 found from my "apt-cache search" before I imported the libpyside2* externally, now I run  "apt-cache policy libpyside2-dev", it outputs:

libpyside2.0:
  Installed: 2.0.0+dev-0~201703151702~rev1755~pkg28~ubuntu16.04.1
  Candidate: 2.0.0+dev-0~201703151702~rev1755~pkg28~ubuntu16.04.1
  Version table:
 *** 2.0.0+dev-0~201703151702~rev1755~pkg28~ubuntu16.04.1 500
        500 http://ppa.launchpad.net/thopiekar/pyside-git/ubuntu xenial/main amd64 Packages
        100 /var/lib/dpkg/status

On the other hand,  I didn't use the /etc/apt/sources.list.d/ros-latest.list aforementioned, so...actually now mine is /etc/apt/sources.list.d/thopiekar-ubuntu-pyside-git-xenial.list with the following content after the PPA repo appended:

deb http://ppa.launchpad.net/thopiekar/pyside-git/ubuntu xenial main
# deb-src http://ppa.launchpad.net/thopiekar/pyside-git/ubuntu xenial main

tfoote, 2017-06-16:
Ok, if you're not using packages.ros.org you are going to have to do a bunch of things either from source or packaged by other parties.

For others if you find this thread add the ROS package repository following the [source installation instructions](http://wiki.ros.org/Installation/Source) and these packages will be available.

lvibb, 2017-06-16:
@tfoote thanks, that's what I suspected. I had to completely clean the source installation and redo ubuntu installation (http://wiki.ros.org/Installation/Ubuntu section 1.2 & 1.3) fixed the issue.

happysf128, 2018-01-23:
@gaoethan Solved mine as well. Thanks.

sypei, 2018-07-05:
@gaoethan Solved mine as well. Thanks.