ros-infrastructure/rosdep #466: Rosdep could not detect OS, tried rosdep update after installation on Ubuntu 16.04

parthchopra1110, 2016-07-06:
rosdep version: 0.11.5

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 545, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 466, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 147, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python2.7/dist-packages/rosdep2/__init__.py", line 85, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.py", line 50, in register_platforms
    register_linaro(context)
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.py", line 64, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 112, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python2.7/dist-packages/rospkg/os_detect.py", line 600, in get_name
    self.detect_os()
  File "/usr/lib/python2.7/dist-packages/rospkg/os_detect.py", line 570, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s"%attempted)
OsNotDetected: Could not detect OS, tried ['windows', 'ubuntu', 'rhel', 'qnx', 'osx', 'opensuse', 'opensuse', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'debian', 'cygwin', 'centos', 'arch']

tfoote, 2016-09-15:
Are you running plain Ubuntu or a derivative OS?

breznak, 2016-09-16:
Looks like duplicate of https://github.com/ros/rosdistro/issues/12716

closed 2017-04-10 by wjwwood