ros-infrastructure/rosdep #473: sudo and rosdep

davetcoleman, 2016-09-14:
When running ``rosdep`` in a Docker container for CI we always get:

```
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
```

Is there a work around?

If not, the debian package ``python-rosdep`` should also require the ``sudo`` package be installed, because currently ``rosdep`` will fail with unuseful error message:
```
rosdep install -r -y -q -n --from-paths . --ignore-src --rosdistro kinetic
ERROR: Rosdep experienced an error: [Errno 2] No such file or directory
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep
rosdep version: 0.11.5
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 353, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 447, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 654, in command_install
    installer.install(uninstalled, **install_options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 485, in install
    verbose=verbose, quiet=quiet)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 535, in install_resolved
    result = subprocess.call(sub_command)
  File "/usr/lib/python2.7/subprocess.py", line 523, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```

That is caused by the Docker container not having ``sudo`` installed by default.

wjwwood, 2016-09-14:
rosdep assumes that sudo is installed, which is the reason you have the "No such file or directory" error. If you like you can add a dependency on sudo in the https://github.com/ros-infrastructure/rosdep/blob/master/stdeb.cfg file and it will be a dependency in the next rosdep release.

The other thing about running `rosdep update` as non-root is different. rosdep assumes that if you're running `rosdep update` as root that you did so with sudo and that it was in error. rosdep is mainly designed to be used as a user, not as root, but since you're running as root always you can safely ignore the warning.

Either way, I don't think these two issues are related, running rosdep as root and rosdep requiring sudo.

tfoote, 2016-09-14:
As @wjwwood mentioned the warning is for most users using sudo if you're knowningly running as root you can ignore the warning. 

I don't think we should add a dependency on sudo. For the use of rosdep without sudo installed in docker you should just use the `--as-root` option to disable the use of sudo.

NikolausDemmel, 2016-09-14:
I implemented a simple possible fix for the sudo issue in #474. Maybe you can give that a shot.

As mentioned by William and Tully the Warning is a separate issue, and it should actually just be a warning, so the update should still work fine, or doesn't it?

davetcoleman, 2016-09-14:
@tfoote what installers would i need to disable? is there a list of all of them? I found this fixes the current issue: ``--as-root apt:false``

wjwwood, 2016-09-14:
@tfoote why not have a dependency on sudo? Just because it is technically optional or some other reason?

Thanks @NikolausDemmel that looks good to me for addressing this particular issue (running rosdep as root in docker without sudo).

However, I think that if we do not add a dependency on sudo, then we should have a better error message when sudo is not available. This exact same issue will come up if @davetcoleman now attempts to rosdep as a non-root user within docker.

NikolausDemmel, 2016-09-15:
@wjwood, considering your comment I'm not sure if this issue is then actually closed by my PR.

After the PR sudo is even "more optional". 

I agree on the need for a better error message though.

tfoote, 2016-09-15:
I'll reopen this to track the better error message. I would be great to detect if it doesn't exist and warn the user if it wants to use it but it's not available. At the same time it can fall back to trying without sudo.

ggoretkin-bdai, 2023-02-15:
Is there a way to suppress `Warning: running 'rosdep update' as root is not recommended.` when running `rosdep` (without `sudo`) in a docker container?

PR #474 (2016-09-14): Installers: Never prepend 'sudo' if already root.
If rosdep is executed as root, we probably never need to prepend a "sudo" to installer commands. This helps in situations where sudo is not installed by default and we typically run as root, eg inside docker images.

Fixes #473.