ros-infrastructure/rosdep #478: Rosdep experiences error "unrecognized version" on macOS Sierra

GlenHenshaw, 2016-10-05:
Running MacOS 10.12.

```aes048076:ros_catkin_ws bash$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"

ERROR: Rosdep experienced an error: unrecognized version: 10.12
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Traceback (most recent call last):
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rosdep2/main.py", line 545, in command_update
    error_handler=update_error_handler)
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rosdep2/sources_list.py", line 466, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rosdep2/gbpdistro_support.py", line 147, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rosdep2/__init__.py", line 85, in create_default_installer_context
    m.register_platforms(context)
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rosdep2/platforms/debian.py", line 50, in register_platforms
    register_linaro(context)
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rosdep2/platforms/debian.py", line 64, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rosdep2/installers.py", line 112, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rospkg/os_detect.py", line 621, in get_name
    self.detect_os()
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rospkg/os_detect.py", line 584, in detect_os
    self._os_codename = os_detector.get_codename()
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rospkg/os_detect.py", line 325, in get_codename
    return _osx_codename(*ver[0:2])
  File "/Users/glenhenshaw/anaconda/lib/python3.5/site-packages/rospkg/os_detect.py", line 304, in _osx_codename
    raise OsNotDetected("unrecognized version: %s.%s"%(major, minor))
rospkg.os_detect.OsNotDetected: unrecognized version: 10.12
```

dirk-thomas, 2016-10-05:
Which version of `rospkg` have you installed? The latest release (1.0.41) should detect *Sierra*.

GlenHenshaw, 2016-10-05:
Yes, sorry - noob problem. I had 1.0.40 installed. It seems to work now.

closed 2016-10-05 by dirk-thomas