ros-infrastructure/rosdep #484: rosdep update  ERROR: Rosdep experienced an error: 'list' object has no attribute 'encode'

menlody-wen, 2016-10-31:
When I finished the sudo rosdep init，putin the “rosdep update”，the problem appeared：
ERROR: Rosdep experienced an error: 'list' object has no attribute 'encode'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.10.27

Traceback (most recent call last):
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 121, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 277, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 285, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/pymodules/python2.7/rosdep2/main.py", line 456, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/pymodules/python2.7/rosdep2/sources_list.py", line 438, in update_sources_list
    retval.append((rds, write_cache_file(sources_cache_dir, dist['distribution'], rosdep_data)))
  File "/usr/lib/pymodules/python2.7/rosdep2/sources_list.py", line 493, in write_cache_file
    key_hash = compute_filename_hash(filename_key)
  File "/usr/lib/pymodules/python2.7/rosdep2/sources_list.py", line 479, in compute_filename_hash
    sha_hash.update(filename_key.encode())
AttributeError: 'list' object has no attribute 'encode'

what is this problem？ How can I slove it？ thank you very much

wjwwood, 2016-11-01:
Please try updating `rosdep`. Version `0.10.27` (which is what your error log says you're using) is quite old, the current version is `0.11.4`.

closed 2016-11-01 by wjwwood

tfoote, 2016-11-01:
Make sure the python `rosdistro` library version is also up to date.