ros-infrastructure/rosdep #486: Homebrew TypeError Mac Python2.7

winzelam1, 2016-11-03:
Received this error when running:
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --verbose

resolution: homebrew [tinyxml, gcc, console_bridge, graphviz]
uninstalled: []
resolution: pip [psutil]
uninstalled: []
resolution: homebrew [poco, cppunit, ffmpeg, yaml-cpp, jpeg, ogre, qhull]
uninstalled: []
resolution: pip [pygraphviz, catkin-pkg]
uninstalled: []
resolution: homebrew [jasper]
uninstalled: []
resolution: pip [rosdep, pydot]
uninstalled: []

ERROR: Rosdep experienced an error: sequence item 0: expected string, HomebrewResolution found
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/Library/Python/2.7/site-packages/rosdep2/main.py", line 353, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/Library/Python/2.7/site-packages/rosdep2/main.py", line 447, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/Library/Python/2.7/site-packages/rosdep2/main.py", line 634, in command_install
    print("uninstalled dependencies are: [%s]"%(', '.join([', '.join(pkg) for pkg in [v for k,v in uninstalled]])))
TypeError: sequence item 0: expected string, HomebrewResolution found

wjwwood, 2016-11-03:
Looks like a bug in the verbose output code. You can work around it by dropping `--verbose` I would guess. However, I don't have time to look into it at the moment.

wjwwood, 2017-07-28:
I believe this should be fixed by https://github.com/ros-infrastructure/rosdep/pull/525

closed 2017-07-28 by wjwwood