ros-infrastructure/rosdep #490: Rosdep failed to detect successful installation of [qt5]

excelulous, 2016-11-24:
Running this command:
`MacBook:ros_catkin_ws user$ rosdep install --from-paths src --ignore-src --rosdistro kinetic -y
executing command [brew install qt5]
Warning: qt5-5.7.0 already installed
ERROR: the following rosdeps failed to install
  homebrew: Failed to detect successful installation of [qt5]`

I also get this stacktrace, but I understand it might be caused by another, unrelated issue:
`rosdep version: 0.11.5

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 353, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 447, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 634, in command_install
    print("uninstalled dependencies are: [%s]"%(', '.join([', '.join(pkg) for pkg in [v for k,v in uninstalled]])))
TypeError: sequence item 0: expected string, HomebrewResolution found
`

I installed pyqt5 separately via this command:
`brew install pyqt5 --with-python`

And Homebrew tells me pyqt5 is installed:
`brew info pyqt5
pyqt5: stable 5.7 (bottled)
Python bindings for v5 of Qt
https://www.riverbankcomputing.com/software/pyqt/download5
/usr/local/Cellar/pyqt5/5.7 (901 files, 42.9M) *
  Built from source on 2016-11-24 at 16:03:51 with: --with-python
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/pyqt5.rb
==> Dependencies
Required: qt5 ✔, sip --with-python3 ✔
==> Requirements
Recommended: python3 ✔
Optional: python ✔
==> Options
--with-debug
	Build with debug symbols
--with-docs
	Install HTML documentation and python examples
--with-python
	Build with python support
--without-python3
	Build without python3 support
`

Any idea what's going on?

tfoote, 2016-11-26:
How do you "also get this stacktrace"? it looks to be a duplicate of #486.

Also can you give the brew info for `qt5` which is the one failing.

excelulous, 2016-11-28:
I ran this command to get the stacktrace:
`rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --verbose`

I managed to get past this by installing qt5 separately with homebrew, and then skipping the qt5 keys like this:
```
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --as-root pip:no --skip-keys="python-qt-bindings-qwt5 libqt5-core libqt5-gui libqt5-opengl libqt5-opengl-dev libqt5-widgets qt5-qmake qtbase5-dev python-imaging"
```

Eventually I was able to install all dependencies.

Here's the brew info for qt5:
```
brew info qt5
qt5: stable 5.7.0 (bottled), HEAD [keg-only]
Version 5 of the Qt framework
https://www.qt.io/
/usr/local/Cellar/qt5/5.7.0 (8,218 files, 326M)
  Poured from bottle on 2016-11-25 at 09:18:50
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/qt5.rb
==> Dependencies
Build: pkg-config ✔
Optional: dbus ✘
==> Requirements
Build: xcode ✔
Required: macOS >= 10.8 ✔
Optional: mysql ✘, postgresql ✘
==> Options
--with-dbus
	Build with dbus support
--with-docs
	Build documentation
--with-examples
	Build examples
--with-mysql
	Build with mysql support
--with-oci
	Build with Oracle OCI plugin
--with-postgresql
	Build with postgresql support
--with-qtwebkit
	Build with QtWebkit module
--without-webengine
	Build without QtWebEngine module
--HEAD
	Install HEAD version
==> Caveats
We agreed to the Qt opensource license for you.
If this is unacceptable you should uninstall.

This formula is keg-only, which means it was not symlinked into /usr/local.

Qt 5 conflicts Qt 4

Generally there are no consequences of this for you. If you build your
own software and it requires this formula, you'll need to add to your
build variables:

    LDFLAGS:  -L/usr/local/opt/qt5/lib
    CPPFLAGS: -I/usr/local/opt/qt5/include
    PKG_CONFIG_PATH: /usr/local/opt/qt5/lib/pkgconfig

.app bundles were installed.
Run `brew linkapps qt5` to symlink these to /Applications.
```

des0ps, 2017-01-19:
I have the same problem, however, using the following command does not work for me:

```
rosdep install --from-paths src --ignore-src --rosdistro indigo -y --as-root pip:no --skip-keys="python-qt-bindings-qwt5 libqt5-core libqt5-gui libqt5-opengl libqt5-opengl-dev libqt5-widgets qt5-qmake qtbase5-dev python-imaging"
```

Any clue on how I could work around it? I am trying to install indigo and I have the same info output for `brew info qt`.

wjwwood, 2017-01-27:
@des0ps I wouldn't use the quotes:

```
rosdep install --from-paths src --ignore-src --rosdistro indigo -y --as-root pip:no --skip-keys python-qt-bindings-qwt5 --skip-keys libqt5-core --skip-keys libqt5-gui --skip-keys libqt5-opengl --skip-keys libqt5-opengl-dev --skip-keys libqt5-widgets --skip-keys qt5-qmake --skip-keys qtbase5-dev --skip-keys python-imaging
```

The option doesn't take arguments as I'd imagine it should, but the above ought to work.

NikolausDemmel, 2017-01-30:
Hmm... not sure what the issue is, but in any case `--skip-keys` should accept space-separated values (https://github.com/ros-infrastructure/rosdep/blob/40067652cd0cde9d192e00788500d67124a30e27/src/rosdep2/main.py#L328)

Maybe there is another rosdep key for you that resolves to brew package qt5, so you might need to add that to the list of skipped keys.

Not sure why the installation of qt5 is not detected, maybe something changed with the options that qt5 installs with by default and what rosdep expects.

folkwang, 2017-03-13:
I've also bumped into this error of `homebrew: Failed to detect successful installation of [qt]`
but `--skip-keys python-qt-bindings-qwt5 --skip-keys libqt5-core --skip-keys libqt5-gui --skip-keys libqt5-opengl --skip-keys libqt5-opengl-dev --skip-keys libqt5-widgets --skip-keys qt5-qmake --skip-keys qtbase5-dev --skip-keys python-imaging` seem no longer to work.
How do you find out the value to plug into --skip-keys to get over it?

dymczykm, 2017-03-17:
``--skip-keys`` did not seem to work for me either, but this is what I did:
* qt5 installation status is not detected because it's a keg only formula
* you can force it get linked to go through the rosdep command by:
  * unlinking the qt4-no-webkit: ``brew unlink qt4-no-webkit``
  * force linking qt5: ``brew link qt5 --force``
* going through the script including the rosdep command
* undoing the changes: unlink qt5, link qt4-no-webkit
* running the catkin build command manually

Maybe this helps.

NataliaDiaz, 2017-04-14:
@dymczykm   brew unlink qt4-no-webkit produces for me:
Error: No such keg: /usr/local/Cellar/qt4-no-webkit

zhonghh, 2017-04-17:
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y
I got:
executing command [brew install qt5]
Warning: qt is a keg-only and another version is linked to opt.
Use `brew install --force` if you want to install this version
ERROR: the following rosdeps failed to install
  homebrew: Failed to detect successful installation of [qt5]

serser, 2017-06-03:
and on macOS sierra 10.12.5, same command with the following error:

    ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:

        webkit_dependency: No definition of [python-qt5-bindings-webkit] for OS [osx]

fferri, 2017-06-20:
Same error here on macOS sierra 10.12.5, and adding `--skip-keys=python-qt5-bindings-webkit` to the command line results in this new error:

```
executing command [brew install qt5]
Warning: qt 5.9.0 is already installed
ERROR: the following rosdeps failed to install
  homebrew: Failed to detect successful installation of [qt5]
```

claudiomenghi, 2017-06-28:
The following command worked for me

`rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --skip-keys "libqt5-core libqt5-gui libqt5-opengl libqt5-opengl-dev libqt5-widgets qt5-qmake qtbase5-dev python-qt5-bindings-webkit"`

Do you know any drawback on this command? I believe that it just does not install some qt5 package which is already installed. Am I correct?

NikolausDemmel, 2017-06-28:
If you have these packages already installed, there is no drawback to using --skip-keys

claudiomenghi, 2017-06-28:
The following command worked for me:

`rosdep install --from-paths src --ignore-src --rosdistro kinetic -y --skip-keys "libqt5-core libqt5-gui libqt5-opengl libqt5-opengl-dev libqt5-widgets qt5-qmake qtbase5-dev python-qt5-bindings-webkit"`

Do you know any drawback on this? I think that I just do not install some qt5 package which is already installed. Am I correct?

pitosalas, 2017-07-19:
I had the same problem too. Does anyone understand the underlying problem? rosdep install reports that brew says that brew failed to detect qt but if you ask brew directly then it is installed... Except for a cryptic message about: 

```
This formula is keg-only, which means it was not symlinked into /usr/local,
because Qt 5 has CMake issues when linked.

If you need to have this software first in your PATH run:
  echo 'set -g fish_user_paths "/usr/local/opt/qt/bin" $fish_user_paths' >> ~/.config/fish/config.fish

For compilers to find this software you may need to set:
    LDFLAGS:  -L/usr/local/opt/qt/lib
    CPPFLAGS: -I/usr/local/opt/qt/include
For pkg-config to find this software you may need to set:
    PKG_CONFIG_PATH: /usr/local/opt/qt/lib/pkgconfig
```

Which I don't understand. I tried adding opt/qt/bin to the path but that didn't solve anything. And once you skip-key on qt5 others prop up.

Also, it's safe if the package (e.g. qt5) is already installed but how do I check that. Especially in the face of a claim that brew says it is not.

NikolausDemmel, 2017-07-20:
> Is it possible that rosdep is looking somewhere else, where brew is not storing?

No, rosdep is parsing the output of the `brew` command and not looking directly for the installed libraries. Maybe it is related to the fact that it is keg-only (and used to be not keg-only), or something in the CLI output of homebrew changed.

> Also, it's safe if the package (e.g. qt5) is already installed but how do I check that. Especially in the face of a claim that brew says it is not.

Not sure what you mean by "is it safe"? If the ROS packages that depend on Qt5 compile, you are good. If during compilation something is missing, you can still try to install it manually using `brew ...` directly.

> Especially in the face of a claim that brew says it is not.

I though homebrew is telling you that it is installed? If not, just install it directly.

pitosalas, 2017-07-20:
I meant "Especially in the face of a claim that rosdep said it was not.

I asked the brew-core people about this and they said this, in case it is useful to you:

> ERROR: the following rosdeps failed to install
> homebrew: Failed to detect successful installation of [qt5]
> Sorry, we have no idea about ROS here, I'd suggest reporting this to them. My guess is that they don't know that brew install qt installs Qt 5 now and need to update their code accordingly.

Do you know exactly how rosdep decides that:

> Got this error message:
> homebrew: Failed to detect successful installation of [pyqt5]

Could it be that it sees the warnings ("this formula is keg only... etc") and decides that it failed to install?

NikolausDemmel, 2017-07-20:
My point is that this should be an issue that is fairly easy to work around. Just make sure the packages are installed (without rosdep) and give the corresponding `--skip-keys` arguments to make sure rosdep can install all the other packages that work correctly. You can even install ROS without ever using `rosdep install ...`. It's just a tool that helps you getting your dependencies installed. Once they are there, no other ROS tool cares what rosdep says (ok, maybe bloom cares, but you're not using that OSX I suspect).

But nonetheless, you are of course right that this issue should be fixed. If someone could have a look and provide a PR, that would be highly appreciated. The code that checks for installation is here: https://github.com/ros-infrastructure/rosdep/blob/2e95fab3d2e04ae735750f627dfb5cfceb0f9c68/src/rosdep2/platforms/osx.py#L172-L220. Indeed it looks like it says "not installed", if the formula is not linked, which keg-only formulas are by definition not after installation.

Note that the fix is not obvious here. Simply accepting unlinked formulas as "installed" is not enough, since you might have a formula that needs to be "linked" in order to work, but which you had previously manually installed and then "unlinked". Maybe rosdep could check if the formula is "keg-only" and then accept it even if it is "unlinked".

Also the error message could of course be improved for the case that the formula is installed, but not linked. However, in rosdep itself there is only "installed" or "not installed", but not (yet) some intermediate state. So you'd have to see how one can pass over this information to the place where the error is actually printed without changing too much in the infrastructure.

And yes, this is surely an issue with rosdep and not Homebrew. I'm just saying that Homebrew is sometimes a bit of a moving target, so once they change certain things, things that used to work with rosdep suddenly stop to work.

pitosalas, 2017-07-20:
Thanks! Looking at the linked code: respect! As usual, everything is far more complex than one thinks even after accounting for knowing this rule. 

Forgive my ignorance: I did see and used the —skip-keys and used it. But what is a “key” in that context? It seemed to me that some of the dependencies are installed as apps in the file system and others are installed as python scripts or header files etc.

For example my most recent error (just posted) is “no module named genmsg”. Other than trying brew (no luck) I don’t know what it is to judge if I can skip it. Is it a key? Any notion of what it is and how I can install it?

Thanks for your patience!

Pito Salas

NikolausDemmel, 2017-07-20:
>  Forgive my ignorance: I did see and used the —skip-keys and used it. But what is a “key” in that context? 

A rosdep key is either a ROS package name, or a key for a 3rd party system dependency (that is not a ROS package). You can find the mapping from rosdep keys to homebrew packages here: https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml

You can also just type something like

```
rosdep db | grep qt5
```

to check what rules there are involving qt5, or if you want to know about a specific rosdep key, you can also do something like:

```
rosdep resolve libqt5-core
```

> It seemed to me that some of the dependencies are installed as apps in the file system and others are installed as python scripts or header files etc.

I'm not sure what you mean by this. What is the difference exactly? Most dependencies are libraries, so they typically provide binaries (shared libs), header files, python libraries, etc... and are installed via homebrew or pip or other package managers depending on the platform. Note that by default homebrew installs everything in `/usr/local/`.

> For example my most recent error (just posted) is “no module named genmsg”. Other than trying brew (no luck) I don’t know what it is to judge if I can skip it. Is it a key? Any notion of what it is and how I can install it?

I haven't checked your other post, but `genmsg` is a ROS package. Those are treated a bit differently and on OSX you cannot install them with rosdep, since you have to get all packages from source and compile them yourself in your catkin workspace (as per the ROS install instructions). To find out about `genmsg` if you don't know, there are multiple things. If it is a ROS package, there should be http://wiki.ros.org/genmsg. You can also check with `rosdep db | grep genmsg` or `rosdep resolve genmsg` or `roslocate describe genmsg`.

wjwwood, 2017-07-28:
I think this can be closed for now, between the suggestions of using `--skip-keys` and upstream changes to how qt is organized in Homebrew.

Feel free to continue the discussion and I can reopen or open a new issue if needed.

closed 2017-07-28 by wjwwood

PHShaw, 2017-08-08:
@NikolausDemmel just tried running `rosdep db | grep qt5` and got the error below:

````
ERROR: Rosdep experienced an error: sequence item 0: expected string, HomebrewResolution found
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/local/lib/python2.7/site-packages/rosdep2/main.py", line 705, in command_db
    resolved_str = " ".join(resolved)
TypeError: sequence item 0: expected string, HomebrewResolution found
````

This can't be helping other issues.  Tried running a `rosdep update` but no change.  Any suggestions?

NikolausDemmel, 2017-08-08:
@PHShaw, this is a bug. You can reproduce it on another OS with `rosdep db --os=osx:10.12` or similar. I opened a separate issue: #540

NikolausDemmel, 2017-08-08:
Btw, just for your reference, after fixing this bug I currently get this output:

```
$ rosdep db --os=osx:10.12 | grep qt5
libqt5-opengl-dev -> qt5
libqt5-opengl -> qt5
libqt5-core -> qt5
python-qt5-bindings-gl -> pyqt5 sip
python-qt5-bindings-webkit -> pyqt5 sip
libqt5-gui -> qt5
libqt5-widgets -> qt5
qtbase5-dev -> qt5
qt5-qmake -> qt5
python-qt5-bindings -> pyqt5 sip
```

kometa-triatlon, 2017-10-07:
```
brew install qt5 pyqt5
cd /usr/local/Cellar
sudo ln -s pyqt pyqt5
sudo ln -s qt qt5
```