ros-infrastructure/rosdep #494: Incorrect dependency install order using manifests

mrath, 2017-01-10:
We recently switched some of our ROS development from a bunch of custom dependency install (bash) scripts to rosdep. Some of them we created proper .rdmanifest files for, including the `depends` tag. 

However, the installation order doesn't work as expected. At the bottom is an example output, from adding an echo statement to each check presence script. We can see a couple of things from this:

 1. The check presence scripts are executed twice, for some reason
 2. The first time, the order is correct (e.g. `rcw_statistics_database` depends on `lamp`) - the second time it's wrong
 3. When installation begins, it uses the wrong order (the second one)

We're using Ubuntu 14.04.5 and Indigo. Let me know if you need more info.

    $ rosdep install -y -i --from-paths /home/builder/RCW_workspaces/rcw_ci_ws/src
    check-presence-script: ros_patches
    ​check-presence-script: lamp
    check-presence-script: mysql-connector
    check-presence-script: rcw_database
    check-presence-script: robwork
    check-presence-script: rcw_statistics_database
    check-presence-script: smile

    (here it is installing one dependency through apt, that has nothing to do with this)

    check-presence-script: mysql-connector
    check-presence-script: smile
    check-presence-script: rcw_statistics_database
    check-presence-script: lamp
    check-presence-script: rcw_database
    check-presence-script: robwork

    executing command [rosdep-source install http://some_url/mysql_connector.rdmanifest]
    executing command [rosdep-source install http://some_url/smile.rdmanifest]
    executing command [rosdep-source install http://some_url/rcw_statistics_database.rdmanifest]​
    
    (installation fails due to missing dependencies)

NikolausDemmel, 2017-01-10:
How do you specify the dependencies between the rosdep keys that define the required order?

wjwwood, 2017-01-10:
Yeah, can you post your rosdep rules file for these custom keys?

mrath, 2017-01-11:
Can't really post them in full here, since everything is behind a firewall, and the install scripts contains some passwords etc. (don't worry, I know I'm not supposed to do that!).
All the dependencies are set with the `depends` flag in the manifest:
- lamp: no dependencies
- rcw_statistics_database: `depends: [lamp ]`
- rcw_database: `depends: [lamp ]`
- mysql-connector: no dependencies
- smile: no dependencies
- robwork: no dependencies
I did some more testing yesterday (found the `-s` flag to rosdep), and determined that the attempted install order in all cases actually is the opposite of what it's supposed to be. It still runs all the `check-presence-script` twice, though, and in different orders.

mrath, 2017-01-11:
Actually, strike that. It's not deterministic. I can run the same simulated install multiple times, and get both correct and incorrect outcomes. It's like the `depends` flag is ignored...?

wjwwood, 2017-01-11:
It sounds like a bug, but it's hard to know if you have it setup correctly. I know it's a lot of work, but perhaps you could anonymize the rosdep file (changing or removing urls and passwords or even key names)?

jlack1987, 2017-07-21:
We are currently running into this as well with some dependencies in our `package.xml` files. We need to PR the rosdep keys causing the issue, but until we do the keys are `libasound2-dev` and `python-rtmidi`. We have the following key defined in our `python.yml`
```
python-rtmidi:
  ubuntu:
    pip:
      depends: [libasound2-dev]
      packages: [python-rtmidi]
```

and in our `base.yml`
```
libasound2-dev:
  ubuntu:
    apt:
      packages: [libasound2-dev]

```

And we get unpredictable behavior when our CI docker runs `rosdep install`. Sometimes it gets them in the correct order and sometimes it doesn't. Any thoughts?

pauldinh, 2017-07-21:
Adding on to what @jlack1987 said above, here's the output of two successive `rosdep install -s -v` calls. The first time, the `libasound2-dev` apt package gets installed before the `python-rtmidi` pip package. The second time, the pip package gets installed first and fails because it doesn't have the missing apt dependency.

```
uninstalled dependencies are: [gcovr, libasound2-dev, uncrustify, cmakelint, doxygen, python3-coverage, cppcheck, lcov, python-rtmidi, pylint, libxml2-utils, cppcheck-junit]
install options: reinstall[False] simulate[True] interactive[False]
install: uninstalled keys are gcovr, libasound2-dev, uncrustify, cmakelint, doxygen, python3-coverage, cppcheck, lcov, python-rtmidi, pylint, libxml2-utils, cppcheck-junit
#[pip] Installation commands:
  pip install -U gcovr
#[apt] Installation commands:
  apt-get install -y libasound2-dev
  apt-get install -y uncrustify
#[pip] Installation commands:
  pip install -U cmakelint
#[apt] Installation commands:
  apt-get install -y doxygen
  apt-get install -y python3-coverage
  apt-get install -y cppcheck
  apt-get install -y lcov
#[pip] Installation commands:
  pip install -U python-rtmidi
  pip install -U pylint
#[apt] Installation commands:
  apt-get install -y libxml2-utils
#[pip] Installation commands:
  pip install -U cppcheck-junit
```

```
  uninstalled dependencies are: [python-rtmidi, doxygen, uncrustify, python3-coverage, libasound2-dev, libxml2-utils, cppcheck, lcov, pylint, cppcheck-junit, cmakelint, gcovr]
install options: reinstall[False] simulate[True] interactive[False]
install: uninstalled keys are python-rtmidi, doxygen, uncrustify, python3-coverage, libasound2-dev, libxml2-utils, cppcheck, lcov, pylint, cppcheck-junit, cmakelint, gcovr
#[pip] Installation commands:
  pip install -U python-rtmidi
#[apt] Installation commands:
  apt-get install -y doxygen
  apt-get install -y uncrustify
  apt-get install -y python3-coverage
  apt-get install -y libasound2-dev
  apt-get install -y libxml2-utils
  apt-get install -y cppcheck
  apt-get install -y lcov
#[pip] Installation commands:
  pip install -U pylint
  pip install -U cppcheck-junit
  pip install -U cmakelint
  pip install -U gcovr
```

wjwwood, 2017-07-22:
I think I fixed this in https://github.com/ros-infrastructure/rosdep/pull/529.

Can anyone give it a try?

You could also try removing all but one reference to the key which has the depends, in @jlack1987's case `python-rtmidi`. That would mean only one package depends on it and only `build_depend` or something like that. I don't recommend it as a work around, but that it just would indicate it's the same behavior I fixed in #529.

jlack1987, 2017-07-22:
Actually I think we do only have a single package depending on `python-rtmidi`. @pauldinh am I correct about that?

jlack1987, 2017-07-24:
@wjwwood how do we test the fix? Do we just clone this repo, checkout `fix_depends` branch,  install it via `python setup.py install` and then run as normal and see if it works?

wjwwood, 2017-07-24:
Yes that will work. You'll probably want to uninstall it with pip afterwards so the apt version stays the primary one.

jlack1987, 2017-07-25:
Can't say with absolute certainty(must not be a unit test for this), but I ran through 5 or 6 iterations of this and they never installed in the incorrect order after installing your `fix_depends` branch, so it's possible this will fix this problem.

closed 2017-07-27 by wjwwood