ros-infrastructure/rosdep #495: Can't use rosdep init with local user

mightyCelu, 2017-01-11:
Currently it is not possible to setup ros as a non-priviledged user, because '''rosdep init''' will always write to /etc/ros. rospkg provides a function '''get_etc_ros_dir''' that by default uses /etc/ros but will use the contents of the environment variable ROS_ETC_DIR if it is set. However, this call is never executed and its default is used instead.

tfoote, 2017-01-11:
I can reproduce: 

```
tfoote@snowman:~ Last: [0] (0s Seconds)
$ ROS_ETC_DIR=/tmp/etc rosdep init
ERROR: default sources list file already exists:
	/etc/ros/rosdep/sources.list.d/20-default.list
Please delete if you wish to re-initialize
tfoote@snowman:~ Last: [1] (0s Seconds)
$ ROSDEP_SOURCE_PATH=/tmp/etc/rosdep rosdep init
ERROR: default sources list file already exists:
	/etc/ros/rosdep/sources.list.d/20-default.list
Please delete if you wish to re-initialize

```
t looks to be a problem here: https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/sources_list.py#L99-L104 since it's usually invoked with sudo it can't get the environment. 

But if you're trying to change where it is using the environment variable to run as non-privleged that won't work. 

Note that we generally strongly recommend using a system wide installation of rosdep otherwise different users can end up with conflicting definitions. And generally when you're using rosdep to install things you need sudo anyway.  Usually the problem is the other way around like #473

allenh1, 2017-09-03:
@tfoote I think I'll take a look into this one

allenh1, 2017-09-27:
Ok, PR is up.

karnikram, 2019-04-30:
`ROSDEP_SOURCE_PATH` no longer works for me.

```bash
$ ROSDEP_SOURCE_PATH=/tmp/etc/rosdep rosdep init
ERROR: cannot create /etc/ros/rosdep/sources.list.d:
	[Errno 13] Permission denied: '/etc/ros'
$ rosdep --version
0.15.1
```

dirk-thomas, 2019-04-30:
> ROSDEP_SOURCE_PATH no longer works for me.

@karnikram You need to invoke `rosdep init` with `sudo` as stated above.

codebot, 2023-01-17:
This is an ancient ticket; I'm just following up with my experience just now, in case it helps anyone. The error message will always print `Permission denied: '/etc/ros'` but that error message is not correct if `ROSDEP_SOURCE_PATH` is specified. It will also print that (misleading) message if the directory specified by `ROSDEP_SOURCE_PATH` does not exist.

If you're using `ROSDEP_SOURCE_PATH` then just make sure the path specified by the env var exists, and then `rosdep` will happily use it.

tfoote, 2023-02-14:
I've confirmed that ROSDEP_SOURCE_PATH is working now for usage as a non-root user. As this is not the recommended usage I'm going to close this as working now. 

The logic to set the source path trims any directory that does not exist. 

https://github.com/ros-infrastructure/rosdep/blob/b6ad1896e89feff9c5328b7baa708187d790d1e1/src/rosdep2/sources_list.py#L86

As such if the new directory doesn't exist it fall back to the default one. Which leads to the permission denied. For the init capability it should not prune in this case.

closed 2023-03-14 by cottsay